dotnet-bluetooth-le/dotnet-bluetooth-le #192: Effectively stopping updates and unsubscribing from ValueUpdated events

Yelnik, 2017-04-20:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Connect to bluetooth device, start receiving values from a characteristic with `StartUpdatesAsync()`  and `App._characteristics[1].ValueUpdated += (o, args) => {
                    ReceiveData(o, args);
                };`
 (The device sends data once per 500 ms) The function that receives the data is on a new page that is loaded with `PushModalAsync`

2. Attempt to stop/unsubscribe from the event with `App._characteristics[1].ValueUpdated -= ReceiveData` and `App._characteristics[1].StopUpdatesAsync();`

3. Reload the page with `PushModalAsync` and resubscribe to the event exactly the same as in step 1. The previous subscription will still be running, and has not terminated the previous page properly or released to previous event subscription. Data will be received twice every 500 ms (both threads running)

## Expected behavior
I expect to be unsubscribed from the updates and the thread associated with that subscription to be released

## Actual behavior
The modal page or thread handling the data receiving is still running somewhere in the app. It appears the subscription to the data being received was never terminated properly

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** 
**1.3.0-alpha1**

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
**iOS: 10.3.1      Android: 7.0**

**Device:** e.g. HTC Sensation /i Phone 7 ...
Samsung Galaxy S7 Edge

xabre, 2017-04-20:
@Yelnik Just to make sure, you are sure that your unsubscribe code is called when you close the page? And also that your subscribe code is not called twice soehow when opening the page?

Yelnik, 2017-04-20:
Yes. The unsubscribing is done in `OnBackButtonPressed` and makes it through all of the code.

The subscribe code is not called twice. In fact, the amount of times the data is received will be exactly equal to the amount of times that code has been called _in total_. So, if the page is opened and closed 3 times, it will begin receiving the data 3 times (due to the other threads still running)

Yelnik, 2017-04-20:
Since I don't have access to thread ID's, this is the best I could do.

In the page constructor I randomize a number between 1 and 5000, and treat that as a "random ID". Now after I dispose these pages, and then request data from the bluetooth device again, I can see all of the previous "ID's" still running through that function to receive data and spitting it into the output window.

The pages even try to run followup functions when certain data is received, after the pages have been disposed.

xabre, 2017-04-22:
You are using a lanbda to subscribe to the value updated event (` ValueUpdated += (o, args) => { ReceiveData(o, args); };`) and then unsbuscribing the actual function delegate `ReceiveData`. You should subscribe directly with your delegate ` ValueUpdated += ReceiveData;` otherways there is no way to unsubscribe an anonymous delgate (your lambda).

Yelnik, 2017-04-22:
`ValueUpdated += ReceiveData;` If I subscribe to the event like this, the function never fires. There is no data received this way.

closed 2017-04-23 by xabre