dotnet-bluetooth-le/dotnet-bluetooth-le #194: Receiving different sets of advertisement data on iOS and Android

Yelnik, 2017-04-24:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

**The Bluetooth device we are using is proprietary so it may not be easy to reproduce this issue**
The Bluetooth module has 2 sets of advertisement data by default which is a battery level and a "Device name", the 3rd one was added manually in the firmware. We are attempting to identify the devices by MAC address, but we cannot access the MAC of the remote device with `CrossBluetoothLE.Current.Adapter`, so the MAC address was added in the advertisement data, but does not appear on iOS.

## Steps to reproduce

1. On an Android device and iOS device, search nearby devices

2. When a device is found, check the `AdvertisementRecords`, for this particular device, on Android we are receiving 3 advertisement records (this is correct) on iOS we only receive 2

## Expected behavior
I expect to see the same advertisement data on iOS and Android

## Actual behavior
1 of our advertisement data sets is missing on iOS (MAC address)

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
1.3.0-alpha1

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
iOS: 10.3.1 
Android: 7.0

**Device:** e.g. HTC Sensation /i Phone 7 ...
Samsung Galaxy S7 Edge
iPhone 5C

xabre, 2017-04-24:
What type of AdvertisementRecordType does Android use for the MAC?  
(see [AdvertisementRecordType.cs](../blob/master/Source/Plugin.BLE.Abstractions/AdvertisementRecord.cs)). I suspect it's not implemented on the iOS side

Yelnik, 2017-04-24:
On Android the type for the MAC address is showing as `Plugin.BLE.Abstractions.AdvertisementRecordType.DeviceAddress`

xabre, 2017-04-25:
Based on my knowledge, iOS supports only the following keys for advertisement data:
[advertisement_data_retrieval_keys](https://developer.apple.com/reference/corebluetooth/cbcentralmanagerdelegate/advertisement_data_retrieval_keys)

Maybe you can check your application output for the following message:
https://github.com/xabre/xamarin-bluetooth-le/blob/master/Source/Plugin.BLE.iOS/Adapter.cs#L348
Maybe we are lucky and can see which key iOS is using if at all one.

Maybe this post can also help:
https://forums.developer.apple.com/thread/8442
Let me know what you find.

xabre, 2017-05-05:
@Yelnik any news on this one?

Yelnik, 2017-05-08:
Not yet no, we'll be looking into this in a few weeks maybe. Some other stuff came up and we're fine testing with the Android version for now. I'll post our solution/what happens here when we make the changes.