dotnet-bluetooth-le/dotnet-bluetooth-le #196: await DisconnectDeviceAsync appears to return before it is fully complete

Yelnik, 2017-04-26:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

This is in the case that a user attempts to connect to a device they are already connected to. We are allowing them to do this because sometimes the timeout on the IDevice state seems too long. I mean that if you connect to a device, then unplug that device, it will still think you're connected in the code-behind for some time.

## Steps to reproduce

1.  Connect to device

2. Disconnect from device `await ActiveBLEDevice.DisconnectDeviceAsync(ActiveDevice);`

3. Connect to the device after the Disconnect call returns

## Expected behavior
I expect the awaited disconnect call to fully complete before returning and allowing the connect call to execute

## Actual behavior
Sometimes it will execute properly, sometimes it will attempt to connect before the disconnect call is complete. We can see this when the console echos `close()` after the call to `ConnectToDeviceAsync`

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
1.3.0-alpha1

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
iOS: 10.3.1
Android: 7.0

**Device:** e.g. HTC Sensation /i Phone 7 ...
Samsung Galaxy S7 Edge
iPhone 5C

xabre, 2017-05-10:
Can you show us your console output?

Yelnik, 2017-05-16:
It looks something like:

```
05-16 11:46:25.400 D/BluetoothGatt( 6566): discoverServices() - device: 31:20:0E:93:54:94
05-16 11:46:28.188 D/BluetoothGatt( 6566): onSearchComplete() = Device=31:20:0E:93:54:94 Status=0
05-16 11:46:28.193 D/BluetoothGatt( 6566): onClientConnParamsChanged() - Device=31:20:0E:93:54:94 interval=39 status=0
05-16 11:46:29.157 D/BluetoothGatt( 6566): onClientConnectionState() - status=22 clientIf=6 device=31:20:0E:93:54:94
05-16 11:46:29.167 D/BluetoothGatt( 6566): close()
05-16 11:46:29.169 D/BluetoothGatt( 6566): unregisterApp() - mClientIf=6
05-16 11:46:35.465 D/BluetoothManager( 6566): getConnectionState()
05-16 11:46:35.465 D/BluetoothManager( 6566): getConnectedDevices
05-16 11:46:40.394 D/BluetoothManager( 6566): getConnectionState()
05-16 11:46:40.394 D/BluetoothManager( 6566): getConnectedDevices
```