dotnet-bluetooth-le/dotnet-bluetooth-le #204: IOS: CBCentralManager implementation of RestoreIdentifier is missing

mariusbloemhof, 2017-05-04:
Hi,

on IOS - we are missing the Implementation of **RestoreIdentifier** 

```
            _centralManager = new CBCentralManager(this, DispatchQueue.CurrentQueue, new **CBCentralInitOptions
            {
                ShowPowerAlert = true,
                RestoreIdentifier = "myapp.indentifier"
            });**
```
Also **WillRestoreState** is required to make background modes behave correctly on IOS.

The class that contains CBCentralManager will need to inherit from CBCentralManagerDelegate for this to work

mariusbloemhof, 2018-03-15:
Hi, any feedback about this enhancement?

pravincar, 2018-06-15:
Hi @smstuebe, I think this is straightforward to implement and is a pretty important feature (for us at least). I'm happy to help if this is not on the radar soon.

JanNepras, 2019-02-04:
@pravincar I believe that we need to reimplement creating CBCentralManager in [BleImplementation.cs](https://github.com/xabre/xamarin-bluetooth-le/blob/master/Source/Plugin.BLE.iOS/BleImplementation.cs). I am not sure if this is enough according to apple documentation [here](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html) you need implement constructor which have DispatchQueue null and delegete set to self. @xabre @smstuebe With some advices I would be happy to implement it.

JanNepras, 2019-02-06:
@xabre @smstuebe I found out that this is implemented and merged on branch 2.0.0 [here](https://github.com/xabre/xamarin-bluetooth-le/pull/314). Would it be possible to release it?

xabre, 2019-08-05:
Release with v2.0.1

closed 2019-08-05 by xabre