dotnet-bluetooth-le/dotnet-bluetooth-le #205: Write long data over BLE (+ than 20 bytes)

joriano, 2017-05-10:
## Steps to reproduce

1. use the BLE explorer application

2. modify to send more than 20 bytes

## Expected behavior
I develop a long data function sender like follow : 

```
public async void sendDataBle(List<List<Byte>> trame)
        {
           Task isSendCompleted = new Task();
            for (Byte numPaquet = 0; numPaquet < trame.Count; numPaquet++)
            {
                isSendCompleted  = WriteToTXCharacteristic(trame.ElementAt(numPaquet).ToArray()));
                while (!isSendCompleted.IsCompleted) ;
            }
        }
```
the List<Byte> in list trame have a maximum of 20 Bytes.

I tried this implementation too : 

```
 public async void sendDataBle(List<List<Byte>> trame)
        {
            for (Byte numPaquet = 0; numPaquet < trame.Count; numPaquet++)
            {
                await WriteToTXCharacteristic(trame.ElementAt(numPaquet).ToArray());
            }
        }
```

this is the writeToTXCharacteristic function : 

```
        public async Task WriteToTXCharacteristic(byte [] dataTX)
        {
            if (_UartTXcharac.CanWrite)
                await _UartTXcharac.WriteAsync(dataTX);

            await Task.Delay(200); //try to get tempo between two characteristic write
        }
```

## Actual behavior

When i send less than or 20Bytes, it's ok, i receive my packet.

When i send data (more than 20Bytes), this function try to send 2 packets.
I receive one of them but after that nothing, i never receive the second.

### Crashlog

There are no crash log

## Configuration

**Version of the Plugin:** 1.2.1

**Platform:** e.g. iOS 10 / Android 4.4

**Device:** e.g. Wiko raimbow 4G

Tijgerd, 2017-10-02:
maybe you can try this:

```
byte[] sendString = //put your bytes here
List<byte[]> sendChucks = new List<byte[]>();
if(sendString.Length > 20)
{
     int totalChucks = (sendString.Length / 20) + 1;
     byte[] foundChunk = new byte[20];
     int i;
     for(i = 0; (i<sendString.Length) && (sendString.Length-i > 20);i=+20)
      {
              Array.Copy(sendString, i, foundChunk, 0, 20);
              sendChucks.Add(foundChunk);
      }
      if(sendString.Length-i > 0)
      {
           byte[] lastChunk = new byte[(sendString.Length - i) + 1];
           Array.Copy(sendString, i, lastChunk, 0, ((sendString.Length - i)));
           sendChucks.Add(lastChunk);
      }
 }
else
{
                sendChucks.Add(sendString);
}
```

after that use a foreach(byte[] sendByte in sendChucks) to write your data

-----------------------------------------------------------------------

but I don't think that this is the best solution.

trix0, 2019-10-03:
Does anybody has solution for this ?

Tijgerd, 2019-10-03:
just divide the message in different chucks, BLE is not intended for long messages

trix0, 2019-10-03:
@Tijgerd I have my data divided into the chunks of data. My question is: How do I mark the start and how do I mark the last chunk.

I am very confused in here, IF I look to this [stackoverflow question](https://stackoverflow.com/questions/24135682/android-sending-data-20-bytes-by-ble), In the answer there is something like 
` initial_packet[0] = BLE.INITIAL_MESSAGE_PACKET;`
Do I need to specify this somehow or just send data randomly and my characteristic should take care of it ?

Tijgerd, 2019-10-03:
I havent seen "INITIAL_MESSAGE_PACKET" at the time I was using Xamarin so I can't help you with that.

If you consider using my method, there are serveral ways to "mark" the chunks:

1 (I used this one and never had any "false-positive" outcomes): 
- In the first part of the message, send the total number of bytes you are going to send
- Send all the chunks
- The last chunk should contain a CRC to check if all previous bytes where send correctly

2:
- Mark the start of each chunk with the chunk-number (this will reduce the maximum data-bytes you can send with each chunk but is more secure)
- You will still need to send the number of chunks you are going to send
- A CRC is still advised

Just be creative,
Hope this helps!