dotnet-bluetooth-le/dotnet-bluetooth-le #208: Crash on AdvertismentData parsing

enemaerke, 2017-05-18:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

Unfortunately I cannot reproduce, but is pasting in a crash dump from our HockeyApp crash logs. Also, looking into the code, the following unguarded as-cast could be the culprit:

`NSData.FromString(advertisementData.ObjectForKey(key) as NSString).ToArray())`

## Expected behavior
End user experienced a crash

## Actual behavior
No crash, rather have incomplete BLE advertisment data received

### Crashlog

If something causes an exception paste full stack trace + Exception here

From HockeyApp

System.ArgumentNullException: Value cannot be null.
System.Runtime.InteropServices.Marshal.copy_from_unmanaged(intptr, int, Array, int)(wrapper managed-to-native)
System.Runtime.InteropServices.Marshal.Copy(IntPtr source, byte[] destination, int startIndex, int length)
Foundation.NSData.ToArray()
Plugin.BLE.iOS.Adapter.ParseAdvertismentData(NSDictionary advertisementData)
at Plugin.BLE.iOS.Adapter.<.ctor>b__6_0 (System.Object sender, CoreBluetooth.CBDiscoveredPeripheralEventArgs e) <0x101960948 + 0x00133> in <e5a9507c202a48be9cfdab3b53123a2c#a7729f03f0e764ea04e16ef90cc68af9>:0
CoreBluetooth.CBCentralManager._CBCentralManagerDelegate.DiscoveredPeripheral(CBCentralManager central, CBPeripheral peripheral, NSDictionary advertisementData, NSNumber RSSI)
UIKit.UIApplication.UIApplicationMain(int, string[], intptr, intptr)(wrapper managed-to-native)
UIKit.UIApplication.Main(string[] args, string principalClassName, string delegateClassName)
BC.Mobile.Application.Main(string[] args)

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
version 1.1.0 via NuGet

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
iOS 10.3.1

**Device:** e.g. HTC Sensation /i Phone 7 ...
iPhone 6

xabre, 2017-05-18:
We probably need to make this more robust:
`NSData.FromString(advertisementData.ObjectForKey(key) as NSString).ToArray())` 
The error states that toArray can not be called on null. Most probalbey `NSData.FromString(advertisementData.ObjectForKey(key) as NSString` is `null` ...

enemaerke, 2017-06-07:
Hi @xabre 
Agreed that this needs to be a little more robust. Can I assist in handling this (we have an app with a number of crashes relating to this)

PR #289 (2018-04-07): Fixed the issue of empty NSData ToArray()
Fixed #208 and #189
closed 2018-04-07 by flyingxu