dotnet-bluetooth-le/dotnet-bluetooth-le #209: Cant install MvvmCross.Forms.Presenter

evgesha14, 2017-05-18:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Create a new Xamarin Forms project (PCL) in VS2017.

2. Try to install MvvmCross.forms.presenter package

3.

The thing is that i was able to install MvvmCross.Core and MvvmCross.Platform without any issues

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

I get an installation error, message bellow	

### Crashlog

Severity	Code	Description	Project	File	Line	Suppression State
Error		Could not install package 'MvvmCross.Forms.Presenter 4.4.0'. You are trying to install this package into a project that targets '.NETPortable,Version=v4.5,Profile=Profile259', but the package does not contain any assembly references or content files that are compatible with that framework. For more information, contact the package author.			0	

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

4.4.0

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

Wanted to build it for Android 4.3 as a first step.

**Device:** e.g. HTC Sensation /i Phone 7 ...

smstuebe, 2017-05-18:
What does it have to do with the plugin? We don't have this package as dependency.
Ask here, if you have an MvvmCross related question/issue: https://github.com/MvvmCross/MvvmCross

evgesha14, 2017-05-18:
I apologize if this was not the right place to ask this question. I just thought that because the sample app uses the MvvmCross.Forms heavily, someone here might know the reason for this error.

smstuebe, 2017-05-18:
Sorry, but we can't support any Xamarin / MvvMCross related issue here. Stackoverflow or the linked MvvMCross github page is a way better resource for your issue.

closed 2017-05-18 by smstuebe