dotnet-bluetooth-le/dotnet-bluetooth-le #211: Characteristic does not support update on Android

vbisbest, 2017-05-22:
I am getting "Characteristic does not support update" only on Android when calling StartUpdatesAsync();  Writes are not working either and I am calling them on the main UI thread.   

on iOS, the same code works great and notifications are recieved.

Details:
Latest stable XF
Plugin BLE 1.3.0 Beta 3
Android Galaxy Tab A
Raspberry Pi with Bluez 5.45 using a GATT service

vbisbest, 2017-05-22:
Screenshot showing the different write properties on the characteristic between iOS and Android.  
[image]

robauer, 2018-03-08:
Any follow up on this? I'm seeing the same issue.

insolito86, 2023-07-20:
I am having the same issue when calling StartUpdatesAsync():

[image]

My code:

[image]

Client: Android 8.1 (API 27), Plugin.BLE 2.1.3, .net 7 maui app (VS2022)
Server: esp32 with BLE_SERVER

In "nRF Connect" app on the same device i can connect, get and set values to a characteristic

Any solution?