dotnet-bluetooth-le/dotnet-bluetooth-le #212: Gatt SetCharacteristicNotification FAILED. (only 2nd attempt)

shayo, 2017-05-25:
## Steps to reproduce

1. connect to ble device (I'm using nrf51822)
2. pick a characteristic. start listening on notification updates.
3. disconnect from the device
4. re-connect to the same device.
5. attempt to start listening on notifications again.

## Expected behavior
notifications should get registered after step [5], just like they do after step [2].

## Actual behavior
notifications are properly registered after step [2], but after step five, the following error appears:
Gatt SetCharacteristicNotification FAILED.

### Crashlog

Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt SetCharacteristicNotification FAILED.
  at Plugin.BLE.Android.Characteristic+<StartUpdatesNativeAsync>d__21.MoveNext () [0x0005c] in C:\Projekte\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:109 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/4468/f913a78a/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x00047] in /Users/builder/data/lanes/4468/f913a78a/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:187 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in /Users/builder/data/lanes/4468/f913a78a/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:156 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in /Users/builder/data/lanes/4468/f913a78a/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:128 
  at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in /Users/builder/data/lanes/4468/f913a78a/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:113 
  at ClimbTag.ClimbTagBluetoothDevice+<PairingProcedure>c__asyncD.MoveNext () [0x0044c] in /Users/shayo/Projects/ClimbTag/ClimbTag/Services/ClimbTagBlueToothDevice.cs:464 

## Configuration

**Version of the Plugin:** e.g. 1.2.3

**Platform:** Only happens on Android (5.0). Works without a problem on iOS.

**Device:** Moto G.

shayo, 2017-05-25:
Destroying the device object, followed by searching, connecting and starting updates works. It's something that has to do with some stored state of the object in memory...

xabre, 2017-06-20:
@shayo Did you by any chance also try it with the latest 1.3.0-beta3. 
We added a fix in [1.3.0-beta.2]
      - #198 Android. Clear cached services, characteristics etc on signal loss

shayo, 2017-06-20:
I will check.

shayo, 2017-07-10:
Did not seem to solve the problem.
I also encounter other issues related to querying for the same service twice (2nd attempt fail).

bed007, 2017-07-19:
Juste creare a PR that solves this issue on 1.3.0 branch

xabre, 2017-08-16:
Is this still an issue in 1.3.0?

closed 2019-08-05 by xabre

vimalaven, 2020-06-04:
shayo, can you provide sample

ygl-rg, 2024-03-23:
I have tested the version 3.0 and the latest beta version and noticed once Gatt SetCharacteristicNotification FAILED was thrown, the ble was malfunction completely, The only way to resume is restarting the App.

I tested against android phones such as Redmi, Samsung and oppo.