dotnet-bluetooth-le/dotnet-bluetooth-le #213: StartUpdatesAsync() throws WriteNotPermitted exception on some devices.

xytec, 2017-05-26:
## Steps to reproduce

1. Connect to device.

2. Get a service.

3. Get a characteristic with Notify property.

4. Assign ValueUpdated eventhandler.

5. StartUpdatesAsync().

## Expected behavior
StartUpdatesAsync() should complete without error.
I've tested the same code on 3 devices, and it worked on 1 of them.
Device info is provided below in the Configuration section.

## Actual behavior
StartUpdatesAsync() throws WriteNotPermitted exception.

### Crashlog

System.Exception: WriteNotPermitted
at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at Plugin.BLE.Abstractions.Utils.TaskBuilder+<FromEvent>d__1`3[TReturn,TEventHandler,TRejectHandler].MoveNext () [0x000d0] in C:\\Users\\Adrian\\Documents\\GitHub\\xamarin-bluetooth-le\\Source\\Plugin.BLE.Abstractions\\Utils\\TaskBuilder.cs:51 \n--- End of stack trace from previous location where exception was thrown ---\n  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at Plugin.BLE.Android.Characteristic+<StartUpdatesNativeAsync>d__21.MoveNext () [0x00262] in C:\\Users\\Adrian\\Documents\\GitHub\\xamarin-bluetooth-le\\Source\\Plugin.BLE.Android\\Characteristic.cs:135 \n--- End of stack trace from previous location where exception was thrown ---\n  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in <d18287e1d683419a8ec3216fd78947b9>:0 \n  at Test.BLE.ViewModels.DeviceListViewModel+<HandleSelectedDevice>d__19.MoveNext () [0x004e6] in E:\\Projects\\Xamarin\\xytec\\Test.BLE\\Test.BLE\\Test.BLE\\ViewModels\\DeviceListViewModel.cs:116

## Configuration

**Version of the Plugin:** 1.2.3

**Platform/Device:**
1. Works on - Android 5.1 (Huawei Y3II "LUA-U22")

2. Doesn't work on - Android 5.1.1 (OnePlus One), Android 4.4.2 (Samsung Mega 6.3 I9205)

FrozenKiwi, 2017-07-29:
I'm seeing the same behaviour, intermittently.

Works on Huaweii Mate 2, Doesnt work on Sony Z5 Compact.

FrozenKiwi, 2017-07-29:
"doesn't work" == "sometimes (75% of the time) throws "WriteNotPermitted" exception when connecting to a Bluno"

CatimusPrime, 2018-05-31:
Hello, 
 I too am experiencing the same issue, but on a device that USED to work. The device is an Acer Iconia 7, the device hosting the notify characteristic is a Feather M0. I can connect to it via NRFConnect and see the values update as I press buttons on the device, but when I try to startUpdatesAsync from my app I receive a "WriteNotPermitted".

Don't know if its related but another characteristic that is written to be a Read, comes in as a Notify (a value it used to be). I have again checked with NRFConnect and the characteristics look correct in that app. Is there some sort of caching going on here?

xabre, 2019-08-05:
Seems to me that writing the CLIENT_CHARACTERISTIC_CONFIG fails. This means that the device manufacturer did not st the correct permission on the descriptor.

The plugin implement starting notifications for Android similar to the docs:
https://developer.android.com/guide/topics/connectivity/bluetooth-le#notification

Please reopen the issue with more details and with regard to the android documentation.

closed 2019-08-05 by xabre