dotnet-bluetooth-le/dotnet-bluetooth-le #215: CancellationTokenSource.Cancel doesn't stop BLE timeout

philsta, 2017-06-06:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Try to connect to known device using
`currentDevice = await BluetoothAdapter.ConnectToKnownDeviceAsync(guid, connectionParameters, tokenSource.Token);`

2. Cancel the establishment of the connection using tokenSource.Cancel()

3. Wait for `05-23 13:45:25.644 D/BluetoothGatt(26827): close()
05-23 13:45:25.644 D/BluetoothGatt(26827): unregisterApp() - mClientIf=5` in debug output

## Expected behavior
If connection establishment is cancelled by the user, it should be immediately possible to connect to establish a new connection.

## Actual behavior
If the connection establishment is cancelled, it takes 20 seconds to establish a new connection (caused by the Android BLE timeout of 20 seconds). Is it possible to cancel that using CancellationToken?

### Crashlog

## Configuration

**Version of the Plugin:** 1.3.0-beta3

**Platform:** Android 6.0.1

**Device:** ASUS Nexus 7

xabre, 2017-06-20:
We have not found a way yet to force android to stop the connection attempt on the spot. It would be very helpful and I would be very interested if by any chance you were able to find it.

As a side note, iOS allows this by calling disconnect if I remember correctly...

xabre, 2017-07-23:
I flagged this as a question and am closing it for now until we find out if it's possible to modify the android timeout.

closed 2017-07-23 by xabre

erdmenchen, 2018-03-21:
That seems related to #275