dotnet-bluetooth-le/dotnet-bluetooth-le #216: Name is coming as null after discovering device

sarkarstanmoy, 2017-06-07:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Used Vanilla version

2. StartScanningForDevicesAsync

3. Once device discovered try to get the name of the device

## Expected behavior
    adapter.DeviceDiscovered +=  (s, e) =>
                {
                    var device = e.Device.NativeDevice;
                   // await e.Device.UpdateRssiAsync();
                    DeviceList.Add(new DeviceDetails()
                    {
                        Name = e.Device.Name.ToString()

                    });

                };
Name should contain device name

## Actual behavior

Name is coming as null

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

Manifest Permission:
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  <uses-permission android:name="android.permission.BLUETOOTH" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />

**Version of the Plugin:** 1.2.3

**Platform:** Android 7 (Nougat)

**Device:** Moto G4 plus

xabre, 2017-06-20:
Does this happen for all BLE devices or just for your specific device? Does a third party BLE scanner app get the name of your device? In case not it might mean that your device is not advertising a name.

GrooverFromHolland, 2017-07-28:
@sarkarstanmoy  this is IMO normal  behavior. I find that discovered devices are listed more than once and sometimes the same device is missing the name or other properties, but there is always a device with complete properties. That is why I always filter the result like this:
`deviceList = new List<IDevice>();
                deviceList = adapter.DiscoveredDevices.Where(x => x.Name != null).Distinct().ToList();`

PawelGasiorowski, 2017-09-05:
I have exactly the same problem. I believe it is up to the device somehow. In my case, the device name on Android is null but the same device name on IOS is correct. This must be a bug inside the library I think.
I tested it on Samsung Galaxy S7.
Can someone help us with this issue?

CBaldovi, 2017-09-12:
I have the same problem and it happens if I have ConnectToKnownDeviceAsync before StartScanningForDevicesAsync

PawelGasiorowski, 2017-09-12:
I have find out why it is happening. The Advertisement record does not contain device friendly name at Android nor IOS. So the natural behavior is that the device name is displayed as “Unknown device” or “N/A”. It works the same way on Android and IOS systems. The difference is that IOS reads device info characteristic when  connecting and store the mac address of the device paired with the info from this characteristic in the system cache. This way when you scan for the devices for the second time, IOS recognizes the device as a known device. I took also IOS device that has never been connected to my app, and then the device name was also empty till I connected the device.

Btw. I was scanning BLE devices before connecting it.

cheers,

CBaldovi, 2017-09-12:
Following with the case I have scanned with StartScanningForDevicesAsync with an Android device that had never been linked with the bluetooth device and if I can see the names, I have tried to reset factory data that removes wifi and bluetooth connections and I can still see the names, provided you do not attempt to connect to a device before scanning. In any case this is not an impediment for me since I am going to connect to the devices through its GUID.
I comment on the case just in case someone helps

a greeting

Hassanmrz, 2018-02-21:
if (action == BluetoothDevice.ActionFound)
                {
                    // Get the BluetoothDevice object from the Intent
                    BluetoothDevice device = (BluetoothDevice)intent.GetParcelableExtra(BluetoothDevice.ExtraDevice);
                    // If it's already paired, skip it, because it's been listed already
                    if (device.BondState != Bond.Bonded)
                    {
                        for (int i = 0; i < DevicesArrayAdapter.Count; i++)
                        {
                            alle_device[i] =DevicesArrayAdapter.GetItem(i);
                        }
                        if (device.Name.Length>1 && !(alle_device.Contains(device.Name + "\n" + device.Address)))
                        DevicesArrayAdapter.Add(device.Name + "\n" + device.Address);

                    }
                    else if (action == BluetoothAdapter.ActionDiscoveryFinished)
                    {
                     
                    }

                }
that's the Solution when the Name not appear bcz it take time ti get the Name from The Bluetooth Device

CBaldovi, 2018-03-05:
Adapter.ScanMode = ScanMode.LowLatency; 

All name visible

Connor2hd, 2018-07-13:
Anyone ever find a solution to this?

steeve49, 2020-06-02:
Same problem here. Tried the sample app and it shows only GUID but not names for all devices.

Rolland-K, 2021-09-28:
any solution?
when I scan device with phone's bluetooth setting, it get other device's name but with BluetoothLeScanner, it return null for devices.
where is solution?

Patafoin, 2021-09-28:
> any solution? when I scan device with phone's bluetooth setting, it get other device's name but with BluetoothLeScanner, it return null for devices. where is solution?

Did you try with classic bluetooth libraries ? For me it was the solution. My device is a specific label printer that is not compatible with BTLe and using classic bluetooth solved the problem (you can find some posts on the internet about that)

dodync, 2021-11-13:
I can get the name with the following code:

```
list = new `List<string>();
adapter.DeviceDiscovered += (s, a) =>
            {
                if (!string.IsNullOrEmpty(a.Device.ToString()) && !list.Contains(a.Device.ToString()))
                {
                    Debug.WriteLine($"FOUND BLE Name: {a.Device.ToString()} - ID: {a.Device.Id}");
                    list.Add(a.Device.ToString());
                }
            };
```