dotnet-bluetooth-le/dotnet-bluetooth-le #218: Registering notifications on reconnection issue - notification handler is called multiple times for one notification from BLE device

leelatoo, 2017-06-16:
Hi, 

Our iOS app connects to a BLE device and registers for a characteristic notification.

## Steps to reproduce

1. Connect app to BLE Device for the first time.
                _adapter.ConnectToDeviceAsync(device, false);
 It connects successfully. App holds the connected device object. 

2. Registers for a characteristic notification. 
                firstCharacteristic.ValueUpdated += FirstCharacteristicValueUpdated;
                await characteristic.StartUpdatesAsync();
It registers successfully and FirstCharacteristicValueUpdated() gets called whenever there is a notification from BLE device.

3. Lets say, BLE device goes out of range and comes back in range of smartphone. In this case, app tries to reconnect to BLE device using the same device object.
             _adapter.ConnectToDeviceAsync(device, false);

4. App registers for the same characteristic notification.
                firstCharacteristic.ValueUpdated += FirstCharacteristicValueUpdated;
                await characteristic.StartUpdatesAsync();
5. Now, if BLE device sends a notification, FirstCharacteristicValueUpdated() gets called twice.

## Expected behavior
One connection/reconnection, when BLE device sends a notification, FirstCharacteristicValueUpdated()  should get called only once.

## Actual behavior
On reconnection to the same BLE device, the FirstCharacteristicValueUpdated() gets called twice when there is any notification from BLE device.
If reconnection happens twice, the FirstCharacteristicValueUpdated() gets called thrice for one notification from BLE device.

Are we doing something wrong here?
Is this the proper way of registering for a characteristic notification on reconnection?

How to clear the notification handler registrations to a characteristic if any, before registering for a notification?

Please advice us asap. 

## Configuration

**Version of the Plugin:** Plugin.BLE.1.2.0-beta3
**Platform:** e.g. iOS 10.2.1
**Device:** e.g. iPhone 6S

alberk8, 2017-06-16:
You need to stop the update as well as remove the ValueUpdated delegate.   Otherwise on the second time round your code will add another ValueUpdated delegate and the thing will fire X number of times the ValueUpdated is added.   

Remember when you add, you need to subtract or dispose.  The code below might not work but the gist is there.

```
_Adapter.DeviceConnectionLost += Adapter_DeviceConnectionLost ;

private void Adapter_DeviceConnectionLost(object sender, DeviceEventArgs e)
{
     _Adapter.StopUpdatesAsync();
     firstCharacteristic.ValueUpdated -= FirstCharacteristicValueUpdated;
}
```

leelatoo, 2017-06-22:
Once device connection is lost, we cannot call this  _Adapter.StopUpdatesAsync() method right?

xabre, 2017-06-23:
Yes that method will not work if not connected, but the event handler subscription has to be removed -=.

leelatoo, 2017-06-27:
Yes, removed the event handler on DeviceConnectionLost. Getting notification only once 👍

But the problem still exists.
 When event handler was not removed on connection lost, the behavior was -
 FirstCharacteristicValueUpdated() got called multiple times per notifcation from BLE device, the last call of FirstCharacteristicValueUpdated() will have the updated characteristic value. So there was a delay in receiving the latest notification.

Removed event handler on connection lost. now FirstCharacteristicValueUpdated() gets called only once with latest characteristic value. But same delay still exists. Can you please advise on this?

leelatoo, 2017-08-01:
Hi,

Can you please advice us on the delay in receiving notifications issue?
It is blocking the application from functioning.

bed007, 2017-08-10:
I've tested connection/notify/unnotify/disconnection cycle using 1.2.3 version several times ( > 10) and I did not notice any delay for the notify. In fact, the BLE device I was using was notifying after a command was write to the characteristic. It was very easy to measure the time between the write and the subsequent notify.

Maybe the delay you are experiencing is due to the BLE device itself and not Plugin.BLE ? Use nRF Connect or BLE Explorer to perform similar routine and see if there is a delay.

nRF Connect allow you to setup macros so you can write a script to test your scenario.

oleg-123, 2017-10-18:
I use version 1.3.0, but this problem can not be solved in this way.