dotnet-bluetooth-le/dotnet-bluetooth-le #222: StartScanningForDevicesAsync throws exception with CancellationToken set on iOS

cipriancoj, 2017-06-27:
Sometimes, like 1 out of 10, when I call:

```
EventHandler onScanTimeoutElapsed = (obj, args) =>
{
     System.Diagnostics.Debug.WriteLine("TIMEOUT SCAN");
};
this.Adapter.ScanTimeoutElapsed += onScanTimeoutElapsed;

var cts = new CancellationTokenSource(3000);
CrossBluetoothLE.Current.Adapter.StartScanningForDevicesAsync(cts.Token).Wait();
```

The method throws an exception, which I can't catch (i added a try/catch around the Scan call without success) but I can see in the Application Output window in VS 2017 for mac.

## Expected behavior
That I get the devices 

## Actual behavior
The app crashes

### Crashlog

```
TIMEOUT SCAN
Unhandled Exception:
System.AggregateException: One or more errors occurred. ---> System.NullReferenceException: Object reference not set to an instance of an object
  at Plugin.BLE.Abstractions.AdapterBase.<StartScanningForDevicesAsync>b__41_1 () [0x00000] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\AdapterBase.cs:60 
  at System.Threading.CancellationToken.ActionToActionObjShunt (System.Object obj) [0x00006] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationToken.cs:173 
  at System.Threading.CancellationCallbackInfo.ExecutionContextCallback (System.Object obj) [0x00007] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:1067 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/executioncontext.cs:957 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/executioncontext.cs:904 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/executioncontext.cs:893 
  at System.Threading.CancellationCallbackInfo.ExecuteCallback () [0x00024] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:1048 
  at System.Threading.CancellationTokenSource.CancellationCallbackCoreWork (System.Threading.CancellationCallbackCoreWorkArguments args) [0x00042] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:898 
  at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers (System.Boolean throwOnFirstException) [0x000c0] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:845 
   --- End of inner exception stack trace ---
  at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers (System.Boolean throwOnFirstException) [0x00133] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:878 
  at System.Threading.CancellationTokenSource.NotifyCancellation (System.Boolean throwOnFirstException) [0x0005f] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:777 
  at System.Threading.CancellationTokenSource.Cancel (System.Boolean throwOnFirstException) [0x00006] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:397 
  at System.Threading.CancellationTokenSource.Cancel () [0x00000] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:362 
  at System.Threading.CancellationTokenSource.TimerCallbackLogic (System.Object obj) [0x0000f] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:526 
  at System.Threading.Timer+Scheduler.TimerCB (System.Object o) [0x00007] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/corlib/System.Threading/Timer.cs:327 
  at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00015] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/threadpool.cs:1279 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/threadpool.cs:856 
  at ObjCRuntime.Runtime.ThreadPoolDispatcher (System.Func`1[TResult] callback) [0x00006] in /Users/builder/data/lanes/4991/104e7b43/source/xamarin-macios/src/ObjCRuntime/Runtime.cs:244 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00009] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/threadpool.cs:1208 
---> (Inner Exception #0) System.NullReferenceException: Object reference not set to an instance of an object
  at Plugin.BLE.Abstractions.AdapterBase.<StartScanningForDevicesAsync>b__41_1 () [0x00000] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\AdapterBase.cs:60 
  at System.Threading.CancellationToken.ActionToActionObjShunt (System.Object obj) [0x00006] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationToken.cs:173 
  at System.Threading.CancellationCallbackInfo.ExecutionContextCallback (System.Object obj) [0x00007] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:1067 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/executioncontext.cs:957 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/executioncontext.cs:904 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/executioncontext.cs:893 
  at System.Threading.CancellationCallbackInfo.ExecuteCallback () [0x00024] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:1048 
  at System.Threading.CancellationTokenSource.CancellationCallbackCoreWork (System.Threading.CancellationCallbackCoreWorkArguments args) [0x00042] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:898 
  at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers (System.Boolean throwOnFirstException) [0x000c0] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.12.0.5/src/mono/mcs/class/referencesource/mscorlib/system/threading/CancellationTokenSource.cs:845 <---

[... cut ...]
2017-06-27 13:29:56.331 TestLocalServeriOS[301:33283] critical:   12  libsystem_pthread.dylib             0x000000018dcf6d94 thread_start + 4
2017-06-27 13:29:56.331 TestLocalServeriOS[301:33283] critical: 
=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================
```

## Configuration

**Version of the Plugin:** 1.3.0 - beta3

**Platform:** iOS 10.2.1 

**Device:** iPhone 6s

xabre, 2017-07-23:
@cipriancoj is this still an issue?

xabre, 2017-08-16:
No news in hte last month. Will consider it as solved. Closing.

closed 2017-08-16 by xabre