dotnet-bluetooth-le/dotnet-bluetooth-le #223: GetSystemConnectedOrPairedDevices throws null argument exception on iOS 10.3

genriquez, 2017-06-27:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

Updated sample to latest Xamarin packages: https://github.com/genriquez/xamarin-bluetooth-le/commits/packageupdate

## Steps to reproduce

1. Call `Adapter.GetSystemConnectedOrPairedDevices()` with no parameters. (DeviceListViewModel.cs from updated Sample)

## Expected behavior
List of devices is returned.

## Actual behavior
ArgumentNullException is thrown

### Crashlog

```
at CoreBluetooth.CBCentralManager.RetrieveConnectedPeripherals (CoreBluetooth.CBUUID[] serviceUUIDs) [0x0004c] in .../source/xamarin-macios/src/build/ios/native/CoreBluetooth/CBCentralManager.g.cs:160 
at Plugin.BLE.iOS.Adapter.GetSystemConnectedOrPairedDevices (System.Guid[] services) [0x00038] in .../xamarin-bluetooth-le/Source/Plugin.BLE.iOS/Adapter.cs:239 
at BLE.Client.ViewModels.DeviceListViewModel.GetSystemConnectedOrPairedDevices () [0x00013] in .../xamarin-bluetooth-le/Source/BLE.Client/BLE.Client/ViewModels/DeviceListViewModel.cs:185 
```
## Configuration

**Version of the Plugin:** master branch

**Platform:** iOS 10.3

**Device:** iPhone 6

robcarmona, 2017-09-15:
@genriquez, I just ran into this issue. The error is due to the missing serviceUUID required when called within iOS (param not needed on android). Seems iOS limits your app from finding all connected and paired peripherals to just the one's you have id's for (security reasons I suppose). 

Reference:
https://github.com/xabre/xamarin-bluetooth-le#get-system-devices

vchelaru, 2024-08-28:
I also ran into this. Ideally this library would tell you that you must specify the UUIDs on iOS instead of letting the exception happen deeper. For anyone else running into it, I did something like this:

```cs
var deviceUuid = deviceInQuestion.Id;

var isConnected = adapter.GetSystemConnectedOrPairedDevices(new Guid[]{deviceUuid}).Contains(deviceInQuestion);
```