dotnet-bluetooth-le/dotnet-bluetooth-le #224: Rapid transmit/receive fails on android, works with iOS

shayo, 2017-07-02:
Main issue:
When executing firmware over the air updates to a nordic NRF51/52 device, there is a stage in which many packets are sent to upload the firmware. Notifications are sent back to indicate packet arrived.
In iOS, everything works fine.
In Android, the system starts to choke and many of the notifications fail to arrive in time, causing the procedure to fail. 
Adding await System.Threading.Tasks.Task.Delay(70) 
between Write steps seem to solve the problem, but significantly slows down the process.

## Steps to reproduce

1. setup a bluetooth device that sends rapid responses (easiest would be to attempt to upload a new firmware to a nordic bootloader)

2. 			
Loop:

await WriteData(dfuCharacteristicPacket, dataPacket, false);
if (!await Task.Run(() => { return (packetReceiptNotificationEvent.WaitOne(packetReceiptNotificationTimeout)); }))
					{
						Debug.WriteLine("Missed packet notification => Procedure failed");
						
					}
					packetReceiptNotificationEvent.Reset();

3.

## Expected behavior
Notifications packets should arrive...
 
## Actual behavior
Packets not arriving unless explicit call to system wait

### Crashlog

## Configuration

**Version of the Plugin:** 
1.3.0

**Platform:** 
 Android 5.1 (Fail) 
iOS  10.3.32 (Works)

**Device:** 
Moto G
iPhone 5S

Tijgerd, 2017-09-27:
I'm having a similar problem, I'm sending something to a custom BLE device (BGM111)
When I look with the android app "Serial Bluetooth Terminal", I immediately receive the correct response.
But the Xamarin app just rarely catches a notification.

Tijgerd, 2017-09-27:
Fixed by delaying the notification with 1000ms.... 
but this is not a real fix, this is just confirming the android application is not able to receive a value quickly after a write..

shayo, 2017-09-27:
Yes. I introduced a small delay as well (I think ~100 ms).
But this severely reduces bandwidth and increases firmware update time (i.e., from ~20 seconds in iOS to > 3 min in Android).

Tijgerd, 2017-09-27:
Jup... frustrating..

Tijgerd, 2017-09-27:
I spy with my little eye:
In "Source/Plugin.BLE.Android/Characteristic.cs"

line 127: await Task.Delay(100);
But this is just for StartUpdate()
When do you call the StartUpdate? before or after the "Write"?

Tijgerd, 2017-09-28:
fixed:
call (and wait for) StartUpdateAsync()
Call it right after connecting with your BLE device and put in a delay right there (the BLE device also needs some time to register your update-request)

after that, only send the StopUpdateAsync() when closing your connection.

I have zero delay right now, the total time between writing and reading is about 10ms :)
The only problem you'll have to solve is how to know if the value you received is in deed the answer you were expecting, I fixed it for my application :+1:

shayo, 2017-09-29:
I don't see any reference to StartUpdateAsync. Where is this implemented?

Tijgerd, 2017-09-29:
In your characteristic ;)
On the mainpage of xabre/xamarin-bluetooth-le search for: Characteristic notifications
Make sure to start this and wait for about 500ms the first time.

Once the BLE device starts sending notifications the "characteristic.ValueUpdated" should trigger as soon as you receive new data from your BLE device

freddy87990, 2018-05-28:
someone will have an example,I only receive once the update of the notify

JasperJian101, 2018-10-24:
Has anyone managed to solve this issue? I'm having the same issue with an NRF52 and Android 6.0. My phone is acting as the client and the NRF52 the server. When I write to the characteristic from the phone, I wait for an ack from the NRF52. The problem comes when the ack responds rapidly and I think the issue is that the sending procedure is still busy so doesn't receive the ack?

JasperJian101, 2019-01-23:
I'm also using the NRF52 as the server and an Android 6.0 as client. I used Write Without Response to the characteristic instead of Write and it solved the issue. The problem occurs when a notification/indication arrives from the Server while a write response hasn't completed.