dotnet-bluetooth-le/dotnet-bluetooth-le #227: StopUpdateAsync does not write the characteristic's descriptor to disable notification

bed007, 2017-07-07:
The Android implementation of StopUpdateAsync is not writing the ClientCharacteristicConfigurationDescriptor to `BluetoothGattDescriptor.DisableNotificationValue` so the peripheral is never told that the notification should be disable.

## Steps to reproduce

1. call StartUpdateAsync() on a ICharacteristic

2. call StopUpdateAsync() on the same ICharacteristic

3. check the peripheral device, it is still performing the notify updates

## Expected behavior
The StopUpdateAsync should, just like the StartUpdateAsync does, write to the ClientCharacteristicConfiguration descriptor to disable the notification.

## Suggested fix
Quick fix using #if __ANDROID__ after my call to StopUpdateAsync:

```
var chr = await GetCharacteristic(device, svcId, chrId);

chr.ValueUpdated -= Characteristic_ValueUpdated;
await chr.StopUpdatesAsync();

#if __ANDROID__
// This is a fix for a bug in Plugin.BLE that does not write Notify Disable in the descriptor
var descriptor = await chr.GetDescriptorAsync(ClientCharacteristicConfigurationDescriptorId);

await descriptor.WriteAsync(Android.Bluetooth.BluetoothGattDescriptor.DisableNotificationValue.ToArray());
#endif
```

## Configuration

**Version of the Plugin:** 1.2.3

**Platform:** Android 7.1 (API Lvl 25)

**Device:** Google Nexus 6P

bed007, 2017-07-07:
hello smsissuechecker, I believe I've followed it very much... I don't agree with your analysis ;-)

smstuebe, 2017-07-07:
its just a 25% confidence :P If you want, you can provide a more intelligent approach: http://smstuebe.de/2017/01/12/github-issue-checker-azure-function/

xabre, 2017-07-23:
Merged PRs :) Thank you.

closed 2017-07-23 by xabre