dotnet-bluetooth-le/dotnet-bluetooth-le #232: What is DeviceState.Limited ?

AnoopChatra, 2017-07-19:
Hi,

What is DeviceState.Limited ??

I am using the following method to connect to a BLE device with isAutoConnect set to true in an android application :
_adapter.ConnectToDeviceAsync(device, new ConnectParameters(isAutoConnect),
reconnectCancelToken.Token);

After successful connection the device state will be DeviceState.Connected but however after the reconnection the device state of the remote BLE device would be "DeviceState.Limited" .

Could you please explain what state it is ?? I could not find any co related state in the native android.

## Expected behavior
Expecting DeviceState.Connected for the reconnected device.

## Actual behavior
DeviceState.Limited is the state of the reconnected device.

## Configuration
**Version of the Plugin:** 1.3.0 beta3
**Platform:** Android 7.1.1
**Device:** Google Pixel

Regards,
Anoop Chatra

xabre, 2017-07-23:
@AnoopChatra autoConnect is not completely implemented on our master/1.3.0 branch. We have opened a separate branch where we are working on this feature called `Android_AutoConnect`. Still in progress... 

DeviceState.Limited is a special state we have in Android where the device is connected to the system but not to our app (i.e. we have no gatt instance). This was introduced for the get system connected devices functionality. 

You are right, the status should not be Limited. You can check out the code in our feature branch and give it a try. Feel free to open a PR if you find anyhting.

axa88, 2023-12-25:
> DeviceState.Limited is a special state we have in Android where the device is connected to the system but not to our app (i.e. we have no gatt instance).

Is there anywhere such explinations for all/more objects could be found?