dotnet-bluetooth-le/dotnet-bluetooth-le #238: stuck on await StopUpdatesAsync, and sometimes also on await disconnectFromDevice

shayo, 2017-07-26:
1. Connect to device. 

2. Get some characteristics. Register for StartUpdatesAsync

3. Call await StopUpdatesAsync or 
await disconnectFromDevice

## Expected behavior
No more updated (or disconnection)

## Actual behavior
App get stuck on await StopUpdatesAsync or on await disconnectFromDevice.
Right now I ended up calling those without the await, but this is not optimal.

### Crashlog
no crash log.
## Configuration

**Version of the Plugin: 1.3.0

**Platform:. iOS 10.3.2 

**Device:(iPhone 5c)

JSchaenzle, 2017-07-26:
I've been using the 1.3.0 branch as well and I ran into a similar issue on iOS. For me, I only saw it hang when calling StopUpdatesAsync and I think it only happened when I called it after the peripheral device had already disconnected from the phone. I added a conditional around my call to StopUpdatesAsync that checked to see if the device was still connected before making the call and I haven't had the issue since. I think the issue is that the task relies on a callback from the OS to complete and in some cases, such as when the device is disconnected, the callback never comes. I thought the OS was supposed to throw an exception in that case but I wasn't seeing that on iOS (Not sure about Android).

JSchaenzle, 2017-09-26:
I just ran into this issue again. I can reproduce it pretty easily by turning bluetooth off on iOS while my app is connected to a device. I have confirmed that iOS does not throw an exception when `SetNotifyValue` is called in that case. It just silently does nothing and this method deadlocks because the OS never produces the callback.