dotnet-bluetooth-le/dotnet-bluetooth-le #239: Feature Request: Pass a computed pairing pin from an Android Device to a BLE Peripheral

jopa7002, 2017-07-26:
Hello, as you know this IS A GREAT LIBRARY!!!! I've been using it a lot and its awesome. I really appreciate what your team is doing. Is there a way to pass a computed pairing pin into the automatic pin pairing connection process for Android? If not can this feature be added? Thank you for your time and the effort your team has put into this library.

## Feature Behavior
A pairing pin is created in a class outside of your library.
The pairing pin can be passed into the following methods a ConnectParameters

ConnectToKnownDeviceAsync (Guid deviceGuid, ConnectParameters connectParameters = default(ConnectParameters), CancellationToken cancellationToken = default(CancellationToken));

xabre, 2017-07-26:
@jopa7002 do you know how this is done with the android native sdk? would help a lot with how fast we can get this done if you could help out with the research

adolgov, 2017-07-26:
From what I remember, pairing/pin is handled by the OS side. I recall that this is true for both iOS and Android (I'm pretty certain of iOS, and only somewhat certain for Android).

In fact, I recall that pairing requests are initiated by the peripheral, at which point the OS should pop-up a dialog and take care of that process outside the control of the app.

jopa7002, 2017-07-27:
@adolgov For iOS the pairing process is handled exclusively by the OS however, in Android the pairing process can be processed by the OS or through the application as explained in [this](https://developer.android.com/guide/topics/connectivity/bluetooth.html) guide from Android. Currently, your library is using the application for an automatic connection as demonstrated in these files. 

[image]

@xabre In the Device.cs from lines 79-119 it seems that this is where the connection is established from the mobile platform to the peripheral device. A quick fix would be to prompt the user for the pin through an Android OS level call as explained in [this](https://stackoverflow.com/questions/44317415/android-bluetooth-low-energy-pairing-show-hide-pin-already-loaded-dialog) stack overflow post. If you still want to keep the pin pairing process to be automatic and not prompt the user [this](https://stackoverflow.com/questions/38055699/programmatically-pairing-with-a-ble-device-on-android-4-4) stack overflow post should give you the necessary native code. If the previous stack overflow post is not sufficient then the following posts give example code and explain Androids intricacies. [1](https://github.com/NordicSemiconductor/Android-nRF-Connect/issues/29) [2](https://stackoverflow.com/questions/27633680/bluetoothdevice-connectgatt-with-transport-parameter)

jopa7002, 2017-07-31:
Will this feature request be addressed? If so how can I be of assistance?

jopa7002, 2017-08-07:
@xabre @adolgov What's the plan, how can I help?

xabre, 2017-08-09:
@jopa7002 Hi thanks for the suggestion. At the moment I'm rather pressed with time so I won't make any promises about implementing it myself. 
Maybe you can start working on the implementation of this feature and open a PR. We can gladly discuss you approach/api design :) And I definatley have time to do a review and test :)

jopa7002, 2017-08-10:
Sure! I've already written the code. How should I send it to you so it can
be tested?

Sincerely;

-Payton Jonson

jopa7002, 2017-08-11:
Hey there, I think I submitted a pull request.... I sill new with GitHub so the request might not be in right place. I have a hypothesis to test the code that I've written however, I'm not sure how incorporate the library that I've downloaded and modified into my Visual Studio Solution. What should I do and what resources would help me with this issue? I have not been able to find a useful resource thus far.

jopa7002, 2017-08-18:
Hey there, I've submitted a pull request. What do I need to do next? I have
also modified a copy of the source code that is on my machine. I'm not sure
how to test it because I have not been able to figure out how to use the
library as a reference. Do you have any recommendation?

Sincerely;

-Payton J.

xabre, 2017-08-29:
Hey @jopa7002 ,
I only saw a pull request for mergeing the Android_AutoConnect branch to master. Is that the one you created?

SeshagiriSriram, 2018-05-12:
Has this feature been merged yet?? I do see a pull request from @jopa7002  but not sure if this has been merged yet. This would be useful to a lot of others.

zedweb, 2022-04-24:
Hello , How can i set pairing PIN in my app code? Some devices need PIN like "1234" or "0000" for connecting. Please help me