dotnet-bluetooth-le/dotnet-bluetooth-le #240: ObjectDisposedException on first conection to bluetooth device

ccanbek, 2017-07-31:
On trying to connect to the bluetooth device, system crashed. This is an inconsistent bug, it only happened once or twice.

## Steps to reproduce

1. Connect to a Bluetooth device

## Expected behavior
It should connect

## Actual behavior
Crashed(Inconsistent)

### Crashlog

System.ObjectDisposedExceptionThe CancellationTokenSource has been disposed.

  at System.Threading.CancellationTokenSource.Cancel (System.Boolean throwOnFirstException) [0x00000] in <4a8a6be0aca34168bd49a6cf6aaa6dab>:0 
  at System.Threading.CancellationTokenSource.Cancel () [0x00000] in <4a8a6be0aca34168bd49a6cf6aaa6dab>:0 
  at Plugin.BLE.Abstractions.ICancellationMasterExtensions.CancelEverything (Plugin.BLE.Abstractions.ICancellationMaster cancellationMaster) [0x0000a] in <2dd5a308e4fe43cdbee5665e004d9324>:0 
  at Plugin.BLE.Abstractions.ICancellationMasterExtensions.CancelEverythingAndReInitialize (Plugin.BLE.Abstractions.ICancellationMaster cancellationMaster) [0x00000] in <2dd5a308e4fe43cdbee5665e004d9324>:0 
  at Plugin.BLE.Abstractions.DeviceBase.ClearServices () [0x00000] in <2dd5a308e4fe43cdbee5665e004d9324>:0 
  at Plugin.BLE.Android.Device.CloseGatt () [0x00018] in <1dca03547a2544928ae0f0edf98e3a73>:0 
  at Plugin.BLE.Android.GattCallback.CloseGattInstances (Android.Bluetooth.BluetoothGatt gatt) [0x00043] in <1dca03547a2544928ae0f0edf98e3a73>:0 
  at Plugin.BLE.Android.GattCallback.OnConnectionStateChange (Android.Bluetooth.BluetoothGatt gatt, Android.Bluetooth.GattStatus status, Android.Bluetooth.ProfileState newState) [0x000c0] in <1dca03547a2544928ae0f0edf98e3a73>:0 
  at Android.Bluetooth.BluetoothGattCallback.n_OnConnectionStateChange_Landroid_bluetooth_BluetoothGatt_II (System.IntPtr jnienv, System.IntPtr native__this, System.IntPtr native_gatt, System.Int32 native_status, System.Int32 native_newState) [0x0000f] in <67c369a930454b2c9a19805fbe808c6c>:0 
  at (wrapper dynamic-method) System.Object:9fdd34b1-7d2a-47e7-8f91-485788f8322b (intptr,intptr,intptr,int,int)

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:** Android 6.0

**Device:** Samsung Galaxy J7

Aresak, 2022-04-20:
This still happens frequently.

Stacktrace

`   CancellationTokenSource.Cancel (System.Boolean throwOnFirstException)

    CancellationTokenSource.Cancel ()

    ICancellationMasterExtensions.CancelEverything (Plugin.BLE.Abstractions.ICancellationMaster cancellationMaster)

    ICancellationMasterExtensions.CancelEverythingAndReInitialize (Plugin.BLE.Abstractions.ICancellationMaster cancellationMaster)

    DeviceBase.ClearServices ()

    Device.CloseGatt ()

    GattCallback.CloseGattInstances (Android.Bluetooth.BluetoothGatt gatt)

    GattCallback.OnConnectionStateChange (Android.Bluetooth.BluetoothGatt gatt, Android.Bluetooth.GattStatus status, Android.Bluetooth.ProfileState newState)

    BluetoothGattCallback.n_OnConnectionStateChange_Landroid_bluetooth_BluetoothGatt_II (System.IntPtr jnienv, System.IntPtr native__this, System.IntPtr native_gatt, System.Int32 native_status, System.Int32 native_newState)

    (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.118(intptr,intptr,intptr,int,int)`

TomasMalcanek, 2022-08-19:
It's not crashing very often, but here is a list of device models and androids it happens on. There are no issues on iOS.

Android 10, 11, 12

Redmi Note 11S
OnePlus 9 5G
POCO M4 Pro
P30 Pro
CPH2021
P20 Pro
ZTE Blade V 2020
X500_EEA
Mi Note 10 Lite
VortexV20