dotnet-bluetooth-le/dotnet-bluetooth-le #241: System.InvalidOperationException: Cannot change ObservableCollection during a CollectionChanged event

shayo, 2017-08-05:
## Steps to reproduce

0. ObservableCollection<Plugin.BLE.Abstractions.Contracts.IDevice> listOfDevices
1. Start scanning (have **many** nearby devices active)
2. OnDeviceDiscovered => 
listOfDevices.Add(newly discovered device)

## Expected behavior
device should be added to the list

## Actual behavior
It happens very very rarely, but we experienced a thread-safe issue where multiple devices are triggering the OnDeviceDiscovered at the same time, and this leads to a crash when multiple threads are trying to Add an item to the observableCollection

### Crashlog
Xamarin caused by: android.runtime.JavaProxyThrowable: System.InvalidOperationException: Cannot change ObservableCollection during a CollectionChanged event.
  at System.Collections.ObjectModel.ObservableCollection`1[T].CheckReentrancy () [0x00038] in <519ce3d790cd4da7925178fa5d92adb2>:0 
  at System.Collections.ObjectModel.ObservableCollection`1[T].InsertItem (System.Int32 index, T item) [0x00000] in <519ce3d790cd4da7925178fa5d92adb2>:0 
  at System.Collections.ObjectModel.Collection`1[T].Add (T item) [0x00023] in <9c58d8958db94d9eab964c344f35226c>:0 
  at ClimbTag.PairingService.<onDeviceDiscoveredDiscoveryMode>m__3 (System.Object sender3, System.EventArgs e3) [0x000b0] in <52e5a07605fe49319c4d001503ff1ee2>:0 
  at (wrapper delegate-invoke) <Module>:invoke_void_object_EventArgs (object,System.EventArgs)
  at ClimbTag.ClimbTagBluetoothDevice+<parseCallback>c__async1.MoveNext () [0x002a8] in <52e5a07605fe49319c4d001503ff1ee2>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in <9c58d8958db94d9eab964c344f35226c>:0 
  at ClimbTag.ClimbTagBluetoothDevice+<AckCallback>c__async3.MoveNext () [0x001e7] in <52e5a07605fe49319c4d001503ff1ee2>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in <9c58d8958db94d9eab964c344f35226c>:0 
  at ClimbTag.ClimbTagBluetoothDevice+<ValueUpdatedCallback>c__async4.MoveNext () [0x00075] in <52e5a07605fe49319c4d001503ff1ee2>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore.<ThrowAsync>m__1 (System.Object state) [0x00000] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (System.Object state) [0x0000e] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x0008d] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x0002a] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00096] in <9c58d8958db94d9eab964c344f35226c>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <9c58d8958db94d9eab964c344f35226c>:0

## Configuration

**Version of the Plugin:** 1.3

**Platform:** Android 7.1

**Device:** LGE Nexus 5X

This is probably more related to the issue that ObservableCollection is not ThreadSafe, but I just wonder what is the proper way of keeping a ListView updated as newly devices are discovered.
What is your recommendation? Is there a thread-safe version of ObservableCollection, or another solution?

philsta, 2017-08-10:
Have you tried making listOfDevices public static?
That should guarantee thread-safety according to the msdn documentation of ObservableCollection.

xabre, 2019-08-11:
Invoke on main thread:

closed 2019-08-11 by xabre