dotnet-bluetooth-le/dotnet-bluetooth-le #244: how to catch - Failed to read characteristic - 133

e3dalive2, 2017-08-16:
sometimes i get this error on xamarin.android- ): [ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentException: Failed to read characteristic - 133
Is it possible to catch and ignore it?
full stack:
[ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentException: Failed to read characteristic - 133
08-16 16:59:21.029 E/mono-rt (17439):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <50e73325239843abb1c421bf5dddc7a9>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Reactive.ExceptionHelpers.ThrowIfNotNull (System.Exception exception) [0x0000d] in <ab76b1b1678341d69f8fc2c1f68d0bf5>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Reactive.Subjects.AsyncSubject`1[T].GetResult () [0x00039] in <d0067ed104ac455987b6feb85f80156b>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at Plugin.BluetoothLE.Extensions+<>c__DisplayClass1_1+<<ReadInterval>b__1>d.MoveNext () [0x00027] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Abstractions\Extensions_Characteristics.cs:42 
08-16 16:59:21.029 E/mono-rt (17439): --- End of stack trace from previous location where exception was thrown ---
08-16 16:59:21.029 E/mono-rt (17439):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_1 (System.Object state) [0x00000] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (System.Object state) [0x00007] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono-rt (17439):   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono    (17439): 
08-16 16:59:21.029 E/mono    (17439): Unhandled Exception:
08-16 16:59:21.029 E/mono    (17439): System.ArgumentException: Failed to read characteristic - 133
08-16 16:59:21.029 E/mono    (17439):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <50e73325239843abb1c421bf5dddc7a9>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Reactive.ExceptionHelpers.ThrowIfNotNull (System.Exception exception) [0x0000d] in <ab76b1b1678341d69f8fc2c1f68d0bf5>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Reactive.Subjects.AsyncSubject`1[T].GetResult () [0x00039] in <d0067ed104ac455987b6feb85f80156b>:0 
08-16 16:59:21.029 E/mono    (17439):   at Plugin.BluetoothLE.Extensions+<>c__DisplayClass1_1+<<ReadInterval>b__1>d.MoveNext () [0x00027] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Abstractions\Extensions_Characteristics.cs:42 
08-16 16:59:21.029 E/mono    (17439): --- End of stack trace from previous location where exception was thrown ---
08-16 16:59:21.029 E/mono    (17439):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_1 (System.Object state) [0x00000] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (System.Object state) [0x00007] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.029 E/mono    (17439):   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.034 E/mono-rt (17439): [ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentException: Failed to read characteristic - 133
08-16 16:59:21.034 E/mono-rt (17439):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <50e73325239843abb1c421bf5dddc7a9>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Reactive.ExceptionHelpers.ThrowIfNotNull (System.Exception exception) [0x0000d] in <ab76b1b1678341d69f8fc2c1f68d0bf5>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Reactive.Subjects.AsyncSubject`1[T].GetResult () [0x00039] in <d0067ed104ac455987b6feb85f80156b>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at Plugin.BluetoothLE.Extensions+<>c__DisplayClass1_1+<<ReadInterval>b__1>d.MoveNext () [0x00027] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Abstractions\Extensions_Characteristics.cs:42 
08-16 16:59:21.034 E/mono-rt (17439): --- End of stack trace from previous location where exception was thrown ---
08-16 16:59:21.034 E/mono-rt (17439):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_1 (System.Object state) [0x00000] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (System.Object state) [0x00007] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.034 E/mono-rt (17439):   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <183b200ee49746d48fc781625979428e>:0 
[... cut ...] 28e>:0 
08-16 16:59:21.041 E/mono-rt (17439):   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.041 E/mono-rt (17439):   at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <183b200ee49746d48fc781625979428e>:0 
08-16 16:59:21.041 E/mono-rt (17439):   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <183b200ee49746d48fc781625979428e>:0

xabre, 2017-08-16:
Any exception thrown by an API async api method call can be caught like this:

``
try
{
     var value = await characteristic.ReadAsync();
}
catch(Exception ex)
{

}
``

Are hou handling it otherwise? HOw are you calling ReadAsync?

e3dalive2, 2017-08-17:
i'm using it like this

JSchaenzle, 2017-08-17:
So you're turning the Task returned from `ReadInterval` into an Observable and subscribing to it. When an exception is thrown in an observable stream, it doesn't make it's way out. It just results in the Observable being torn down. The Exception gets propagated via the OnError mechanism similar to how values are sent down the stream with OnNext. If you want to catch/handle the exception you'll need to add a [Catch](https://msdn.microsoft.com/en-us/library/hh211765(v=vs.103).aspx) block before your subscription. Or you could add a `.Retry` which will just ignore the error and try again.

Personally, I would not use the `.Subscribe` call.  I would do something like 
```
try
{
     var value = await characteristic.ReadAsync(); //or any other async api method
     var temp = ParseTemp(value);
     ...
}
catch(Exception ex)
{

}
```

e3dalive2, 2017-08-17:
ok, thank you so much, i will try it