dotnet-bluetooth-le/dotnet-bluetooth-le #246: Multiple connection and BLE notification callback in android on re-connection

AnoopChatra, 2017-08-23:
The application is getting multiple connection call backs and redundant notifications after the re-connections.

## Steps to reproduce

1. Connect a BLE device with android application using  :
_adapter.ConnectToDeviceAsync(device, new ConnectParameters(true),
                                reconnectCancelToken.Token);
I have set the auto-connect parameter to true since i need the automatic re connection. Connection is successful, I will receive only one "BLE device connected" callback. All fine here. 

2.Take the BLE device out of the bluetooth range. The BLE device disconnects from the app I receive only one "BLE device disconnected" callback . All fine here as well.

3.Bring the BLE device back in  bluetooth range. App would automatically reconnects but I am getting 2
"BLE device connected" callbacks.

4. Perform the step 2 and 3 for 'n' times then  I am getting 'n' number of  "connected  and disconnected" callbacks. Multiple callback issue is observed in all sort of re connections.

5. The same behavior is also observed with BLE notification callbacks.  

I did some sort of debugging in Plugin.BLE.Android and I have some observations regarding the root cause of this problem -
1. _gattCallback is one of the attribute of 'Device' class and it is been initialized in the constructor. 

2.  _gattCallback is basically an instance of 'GattCallBack'  which implements BluetoothGattCallback . So this instance contains the connection callback from android OS .

3.  _gattCallback  should have initialized only once but  when ever IDevice object is created new  _gattCallback  is getting created . usually in re-connection IDevice objects are initialized again. 

4. I believe this is the reason for multiple connection callback.

5. We have the exact same problrm with BLE notification callbacks. Multiple notifications are revceived after reconnections.

6. The BLE notification callbacks are also implemented in GattCallBack class(Ideally this call back should have implemented in a different class). So I believe this is the reason for multiple notification callback .

## Expected behavior
Application should receive only one set of connection callbacks and only one set of BLE notifications.

### Crashlog

## Configuration

**Version of the Plugin:** 1.3.0 (Master branch)

**Platform:**  Android 7.1

**Device:** Google Pixel

xabre, 2017-08-29:
@AnoopChatra You are right, this might happen. I have started working on the android auto connect #182 on a separate branch, but some stuff got in the way and I didn't get enough time to continue. This feature will only work reliably after the Android_AutoConect branch is implemented and merged.

AnoopChatra, 2017-09-07:
@xabre , when can we expect the stable release of  'Android_AutoConect branch' ?? We are highly dependent on Plugin BLE library.  Could you please look in to the problem as soon as possible.

xabre, 2017-09-17:
@AnoopChatra We will get to it as soon as we can, this is still an open source project for which we use our spare time to work on. If you can find an improvement/fix for this issue we can gladly merge you pull request :) Thanks

AnoopChatra, 2018-01-18:
@xabre , Is there any updates on the issue ?

DeanZhuo, 2025-08-14:
any news on this issue? August 2025 and still getting multiple events callback.

Jay-57blocks, 2026-02-28:
> any news on this issue? August 2025 and still getting multiple events callback.

Hi, have you found a solution yet?

Are there any reliable solutions for "reconnecting after disconnecting Bluetooth devices"? Implementing stable reconnection logic on Android isn't easy—I'd appreciate your suggestions. thanks.