dotnet-bluetooth-le/dotnet-bluetooth-le #252: ConnectToDeviceAsync not returning or throwing exception when Bluetooth device is powered off

Yelnik, 2017-09-20:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Attempt to connect on iOS to Bluetooth device that is powered off (await ActiveBLEDevice.ConnectToDeviceAsync(n.Device);)

## Expected behavior
I am expecting the call to timeout and likely throw a DeviceConnectionException as it does on Android

## Actual behavior
The call never returns or throws an exception

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:**  iOS 10.3.3

**Device:**   iPhone 5C

xabre, 2017-09-22:
@Yelnik This is actually de default iOS functionality. When your device comes online (even if after hours or days), and you haven't cancelled the connection attempt it will get connected). Android by default will timeout. It's difficult to unify, this will mean we should introduce a synthetic cancellation on ios to match the android behavior. 
A workaround would be to use the cancellation token to trigger the timeout for iOS.

Yelnik, 2017-09-25:
Ok thanks, the cancellation token works in this case then

closed 2019-08-05 by xabre

georgemichailou, 2020-04-03:
I have the same problem on iOS. How can i fix this problem ?