dotnet-bluetooth-le/dotnet-bluetooth-le #254: Characteristic WriteAsync method doesn't throw any exception and jumps at the end of the method

JanNepras, 2017-09-22:
The bug is described bellow. I cloned your repo and in class CharacteristicBase changed method WriteAsync this line:
            var writeType = GetWriteType();
to this:
            var writeType = CharacteristicWriteType.WithoutResponse;

and it started working. If I change the write type to WriteWithoutException outside nuget package it throws the Invalid operation exception. 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Connect to device

2. Retrieve any writable characteristic

3. Call WriteAsync method 

## Expected behavior
WriteAsync method should continue executing rest of the code or throws exception.

## Actual behavior

WriteAsync method jumps at the end of the method even if there is code to execute. This is without throwing any exception.

### Crashlog

No crashlog.

## Configuration

**Version of the Plugin:** 1.3.0 

**Platform:** Android 6

**Device:** Samsung S5

Spargue, 2017-10-09:
Having the same issue. used version 1.1.0 on a nexus 7 (2012) api 5.1 no issues. using it on android 7.0 either 1.1.0 or 1.3.0 produces same results. await characteristic.WriteAsync(..) never returns. Tried compiling against 5.1, 6.0 & 7.0 APIs. Same results. Also tested on a HTC M8 running 5.0.1, works fine.

# Configuration

Version of the Plugin: 1.3.0

Platform: Android 7.0

Device: Samsung Galaxy Tab A

Palku, 2017-11-02:
We also have the same issue. Lenovo A10 Rev2.

ricklz, 2018-02-01:
I am also experiencing this same issue.

Version of the Plugin: 1.3.0
Platform: Android 8.0.0
Device: Google Pixel

xabre, 2019-08-05:
BAsed on what I see here the GATT server wrongly set the characteristic properties to Write = 8 instead of WriteWithoutReponse = 4, so the client tries to send a write with respsonse and is stuck...