dotnet-bluetooth-le/dotnet-bluetooth-le #255: Android: StartScanningForDevicesAsync, some BLE devices are missing

AlexusBlack, 2017-09-26:
It seems that by some reason on only some Android phones some BLE devices are non-searchable, while they are visible on iOS devices without any trouble. 

## Steps to reproduce

1. Bluetooth.StartScanningForDevicesAsync(new Guid[] { new Guid ("0000fff0-0000-1000-8000-00805f9b34fb") });

2. Check on iOS, shows 2 devices with this service Guid

3. Check on Android, shows 1 device with this service Guid

## Expected behavior
Work same and detect all of devices with the service Guid on Android.

## Actual behavior
Android ignore some of devices during scan for devices, they are also non-visible without Guid filter. 

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:** Android 7.0, iOS 11.0

**Device:** Samsung s7 edge, iPhone 7

AlexusBlack, 2017-09-26:
Actually I observe same problem with BLE Scanner, so it should be or OS related (Android by Samsung) or some hardware issue about advertising, not sure, still investigating.

madareklaw, 2017-10-09:
Are you using Android 5.X?
I've had issues similar to what you're describing, it was something to do with the advertising of the BT device

AlexusBlack, 2017-10-09:
@madareklaw so far I seen problem on android 7.0, but yes, so far my theory that this is somehow related to how BLE devices advertise themselve. Thing is that this just started recently and before all these devices where discoverable from same phones...

closed 2017-10-26 by AlexusBlack