dotnet-bluetooth-le/dotnet-bluetooth-le #256: Help/Guide Required for continuous reading of advertised data from non-connectable BLE device.  (Not a plugin issue)

mahesh-it, 2017-10-09:
To @xabre , @smstuebe  and all contributors, thank you very much for this wonderful BLE plugin
form Xamarin forms.   

i apologies in advance for long post.

I have used this plugin in my project to scan, connect BLE device and read required characteristics and it works very well.  

as i mentioned in the subject, my new requirement is  read advertisement records from a non connectable BLE device. the device will advertise data at any point of time (no fixed time) and this requires my app to  continuously scan for advertised data. the information that i need to read is 
available in "AdvertisementRecords"  of a discovered device with type "ManufacturerSpecificData".

i read and i am aware that continuous scanning is not good for battery but that is acceptable in my case. 

following is what i did and problem that i am facing.

**case 1:**  
- Start scan with default scan timeout (10s) with ScanMode=LowPower, 
- After scan starts, i am turning on my BLE device that sends initial advertisement record automatically.
- Scan discovers  device (very quickly in less than a sec)  and i get required data from AdvertisementRecords.
- When i send further advertisement data from BLE device (through available buttons on the device), scan process does not detect new advertised data at all for rest of the remaining scan time.
  
**case 2:**  
- I increased Scan timeout to 1 min start the scan and than start ble device.
- same result as in case 1, only first auto advertised data packet is detected, all following advertised data packets not scanned at all.

**case 3:**  
- i decreased scan time out to only 500ms and scanned in a loop (restarted scan from scan time out callback) 
- scan result improved. now i was able to scan advertised data packet that were sent after BLE device start.
- i would loose certain advertised data packets randomly.

**case 4:**
- i decreased scan time out to only 100ms and scanned in a loop
- scan results improved drastically, i would almost get all advertised data packets with random loses. (may be 2 loss in 10 advertised data packets)

**case 5:**
- i decreased scan time out to only 50ms and scanned in a loop
-  scan did not detect any advertised data packet.

What i want to understand is ,

- is there a reliable way to have continuous scan for advertised data from a non connectable BLE device?

- when scan time out is 1 min, what could be the reason that plugin wont detect for all advertised data packets during scan?

- if i can't scan in a loop,  are there any alternatives to achieve  what i need?

following is what my current code looks like:
`protected override void OnStart()
        {
            ScanAdvertisementData();
        }

        private async void ScanAdvertisementData()
        {
            CrossBluetoothLE.Current.Adapter.ScanTimeoutElapsed += ScanTimeoutElapsed;
            CrossBluetoothLE.Current.Adapter.DeviceDiscovered += DeviceDiscovered;

            CancellationTokenSource tokenSource = new CancellationTokenSource();
            CrossBluetoothLE.Current.Adapter.ScanMode = ScanMode.LowPower;
            CrossBluetoothLE.Current.Adapter.ScanTimeout = 100;
            await CrossBluetoothLE.Current.Adapter.StartScanningForDevicesAsync(tokenSource.Token);
        }

        private void DeviceDiscovered(object sender, DeviceEventArgs e)
        {
            IDevice bleDevice = e.Device;
            AdvertisementRecord advertisedRecord = bleDevice.AdvertisementRecords.SingleOrDefault(ar => ar.Type == AdvertisementRecordType.ManufacturerSpecificData);
            if (advertisedRecord != null)
            {
                MessagingCenter.Send<App, string>(this, "ContractionReceived", advertisedRecord.Data);
            }
        }

        private void ScanTimeoutElapsed(object sender, EventArgs e)
        {
            ScanAdvertisementData();
        }`

i will really appreciate any help/guide related to continuous scanning advertisement records from a non-connectable device.

matC82, 2018-11-14:
Hi mahesh,

you have to use the "DeviceAdvertised" event after you discovered a device to get further advertising data.
The device discovered event will only fired once if a new device is in range.

xabre, 2019-08-11:
ToDo for me:
- [ ] Check if advertisement records are updated for each platform when DeviceAdvertised is raised
- [ ] Update documentation