dotnet-bluetooth-le/dotnet-bluetooth-le #260: Urgent: Scan doesn't work on Samsung S7

shayo, 2017-10-25:
## Steps to reproduce

1. Get a Samsung S7
2. Checkout master branch 
3. Try to scan for devices.

I have an app that runs smoothly on other android devices, but when deployed to Samsung S7, for some reason, it complains that these ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission are not set (which they do).
My manifest clearly says:
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />

## Expected behavior
Near by devices should start appearing 

## Actual behavior
Devices do not show up.
### Crashlog

[BluetoothLeScanner] Start Scan
[BluetoothAdapter] STATE_ON
[BluetoothAdapter] STATE_ON
[BluetoothAdapter] STATE_ON
[BluetoothAdapter] STATE_ON
[BluetoothLeScanner] onClientRegistered() - status=0 clientIf=4 mClientIf=0
[Binder] Binder call failed.
[Binder] java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
[Binder] 	at android.os.Parcel.readException(Parcel.java:1693)
[Binder] 	at android.os.Parcel.readException(Parcel.java:1646)
[Binder] 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:796)
[Binder] 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:423)
[Binder] 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
[Binder] 	at android.os.Binder.execTransact(Binder.java:573)
Thread finished: <Thread Pool> #11

## Configuration

Latest checked from master branch

**Platform:** 
Android 7.0, model: SM-G930T, Kernel: 3.18.31-12142063
**Device:**
Samsung Galaxy S7

shayo, 2017-10-25:
After doing some digging. Can this be related to asking specifically for the user permission?
see https://developer.android.com/training/permissions/requesting.html
?

shayo, 2017-10-25:
Yes. After manually going into the Apps->[My App]->Permissions->Allow all
scanner finds the devices and everything works fine.

Is there a patch you can make to add this feature (i.e., manually asking the user for permission if android version is >= M) ?

shayo, 2017-10-25:
I ended up using the Permissions Plugin in the meanwhile to check if it was set correctly or not.
(https://github.com/jamesmontemagno/PermissionsPlugin).

Please update the documentation. It is no longer sufficient to just modify the manifest file. You may need to actively ask the user for the location permission to use the bluetooth.

xabre, 2017-10-26:
@shayo I'm happy you solved your issue. But actually the following is a quote from our documentation:

> For Marshmallow and above, please follow Requesting Runtime Permissions in Android Marshmallow and don't forget to prompt the user for the location permission.

closed 2017-10-26 by xabre