dotnet-bluetooth-le/dotnet-bluetooth-le #261: GattCallback error 133 when trying to connect via Samsung Galaxy S8+

rcbolt3, 2017-10-26:
Just to clarify, I just started working with Xamarin Forms less than a month ago and was handed a nearly finished product for iOS and was told to basically convert it to Android. So here I am trying to use your plugin (which is what was used successfully for the iOS version) and it works most of the time. My app tries to connect to 4 separate BLE devices and has successfully done so with a Samsung Galaxy Tablet and my old Samsung Note 3 (small issue when connecting to the scale, but that's something entirely different). However, when I use my S8+, I can connect to 2 of the devices just fine, but for some reason the PC-100 and PC-300 devices are giving it fits. Any ideas would be greatly appreciated. Thanks!

## Steps to reproduce
1. Get a Galaxy S8+ and PC-100 and/or PC-300

2. Attempt to ConnectToDeviceAsync

3. Failure?

## Expected behavior
It should succeed.

## Actual behavior
It gives me a Gatt error of status 133.

### Crashlog
Doesn't necessarily crash, but here's the Application output:
[BluetoothAdapter] STATE_ON
[BluetoothLeScanner] Stop Scan
BLE state : On
[BluetoothManager] getConnectionState()
[BluetoothManager] getConnectedDevices
[BluetoothGatt] connect() - device: 88:1B:99:05:1E:14, auto: false
[BluetoothAdapter] isSecureModeEnabled
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=978a772e-bdaa-4d50-bc78-1059283016c4
[BluetoothGatt] onClientRegistered() - status=0 clientIf=5
[BluetoothGatt] onClientConnectionState() - status=133 clientIf=5 device=88:1B:99:05:1E:14
Thread started:  #9
[BluetoothGatt] close()
[BluetoothGatt] unregisterApp() - mClientIf=5
BLE connect DCE ex msg: GattCallback error: 133
e.Device.Name = PC_300SNT
[SemDesktopModeManager] registerListener: android.view.ViewRootImpl$3@1a9318d
[ViewRootImpl@2313f42[PC_300SNT]] ThreadedRenderer.create() translucent=true
[InputTransport] Input channel constructed: fd=142
[ViewRootImpl@2313f42[PC_300SNT]] setView = DecorView@f4a3b53[PC_300SNT] touchMode=true
[ViewRootImpl@2313f42[PC_300SNT]] dispatchAttachedToWindow
[ViewRootImpl@2313f42[PC_300SNT]] Relayout returned: oldFrame=[0,0][0,0] newFrame=[27,829][1053,1328] result=0x27 surface={isValid=true 547435544576} surfaceGenerationChanged=true
[ViewRootImpl@2313f42[PC_300SNT]] mHardwareRenderer.initialize() mSurface={isValid=true 547435544576} hwInitialized=true
[ViewRootImpl@2313f42[PC_300SNT]] MSG_WINDOW_FOCUS_CHANGED 1
[ViewRootImpl@2313f42[PC_300SNT]] mHardwareRenderer.initializeIfNeeded()#2 mSurface={isValid=true 547435544576}
[ViewRootImpl@2313f42[PC_300SNT]] MSG_RESIZED_REPORT: frame=Rect(27, 829 - 1053, 1328) ci=Rect(0, 0 - 0, 0) vi=Rect(0, 0 - 0, 0) or=1
[ViewRootImpl@992761a[MainActivity]] MSG_WINDOW_FOCUS_CHANGED 0
[ViewRootImpl@2313f42[PC_300SNT]] ViewPostImeInputStage processPointer 0
[ViewRootImpl@2313f42[PC_300SNT]] ViewPostImeInputStage processPointer 1
[ViewRootImpl@2313f42[PC_300SNT]] mHardwareRenderer.destroy()#4
[ViewRootImpl@2313f42[PC_300SNT]] dispatchDetachedFromWindow
[InputTransport] Input channel destroyed: fd=142
[SemDesktopModeManager] unregisterListener: android.view.ViewRootImpl$3@1a9318d
[ViewRootImpl@992761a[MainActivity]] MSG_WINDOW_FOCUS_CHANGED 1
[ViewRootImpl@992761a[MainActivity]] mHardwareRenderer.initializeIfNeeded()#2 mSurface={isValid=true 547455545344}
[ViewRootImpl] sendUserActionEvent() returned.

## Configuration

**Version of the Plugin:** 
v1.3.0

**Platform:** 
Android 7.0 

**Device:** 
Galaxy S8+

rcbolt3, 2017-10-30:
Can anybody help me out with this???

JSchaenzle, 2017-10-30:
The Android bluetooth stack does not like to perform multiple operations at the same time. I’ve found that things work much better if you setup your application so that the bluetooth operations are handled synchronously. For example, if you’re connecting to multiple devices, don’t attempt to form those connections at the same time; connect one at a time. Also, disable scanning before you try to connect if possible.

I’ve found that Android devices just tend to produce GATT errors much more often then iOS. Make sure you architect your app to elegantly handle such errors. Also, try getting onto the main thread before performing bluetooth operations. Use Device.BeginInvokeOnMainThread. I wrote this little Task extension that you might find useful…

    public static async Task OnMainThreadAsync(Func<Task> action)
    {
      var tcs = new TaskCompletionSource<object>();
      Xamarin.Forms.Device.BeginInvokeOnMainThread(async () => {
        try {
          await action();
          tcs.SetResult(new { });
        } catch (Exception ex) {
          tcs.SetException(ex);
        }
      });
      await tcs.Task;
    }  

Finally, if you’re having trouble with specific devices, do some searching to see if others are having the same problem (outside of the Xamarin ecosystem). There are known issues with some phones + OS versions.

Good luck!

rcbolt3, 2017-10-30:
Here's the code I'm using to connect:
    `Xamarin.Forms.Device.BeginInvokeOnMainThread(new Action(async () =>`
				`{`
					`try`
					`{`
						`await CrossBluetoothLE.Current.Adapter.ConnectToDeviceAsync(e.Device);`
					`}`
					`catch (DeviceConnectionException ex)`
					`{`
						`Debug.WriteLine("BLE connect DCE ex msg: " + ex.Message);`
                        `Debug.WriteLine("e.Device.Name = " + e.Device.Name);`
                       ` //send error message to screen if callback error 133 = "GattCallback error: 133"`
                       ` SendConnError(e.Device.Name);`
				`	}`
					`catch (Exception ex)`
					`{`
						`Debug.WriteLine("BLE connect basic ex msg: " + ex.Message);`
				`	}`

				`}));`

I'm also making sure that the scanner is stopped before connecting as you can see from the log I posted. And I also wanted to point out that the sample app that you guys suggest to use is also giving the same error when trying to connect to those devices from my phone. I was just wanting to know if there was any solution at all or if I needed to find another BLE library, because I've got it working with a ble.net library I found, but it would require me changing this entire application which is a huge hassle.

JSchaenzle, 2017-10-30:
Hmm. That is odd. The GATT error comes from the OS, not the library, so it’s strange that ble.app works fine. Since there is not a lot of code involved in just forming the connection, I would step line-by-line through each example app and see what the difference is. Both should be calling the same underlying Connect method. If you find anything out please update us. I’m very curious what the difference is.

rcbolt3, 2017-10-31:
So I can't seem to find where they actually make the ConnectGatt() call. If you guys want to look into it, be my guest. It is definitely working with all my devices though. Here's the github repo https://github.com/nexussays/ble.net/tree/master/src/ble.net

mariusbloemhof, 2018-03-15:
GAT 133 errors is well-known because of the synchronous nature of the Android BLE stack. This mostly happens because the device you are trying to connect to is out of range. You can set **ConnectParameters.AutoConnect = true**, which will effectively scan first to make sure the device is in range. Alternatively, just persist the connection attempts when you get GAT 133 errors

samosim, 2018-10-14:
I had a similar problem. Maeiusbloemhof you save my life with a connectParameter hit. Now it works perfectly.
Thanks for help.

SOliasS, 2020-10-19:
I found a Solution, 
i also had to forceBleTransport.
```
 ConnectParameters para = new ConnectParameters(true, true);
                await currentAdapter.ConnectToDeviceAsync(device, para);
```

asadullahqmr, 2020-11-13:
Thanks to the suggestion from @JSchaenzle , disabling scanning before connecting resolved this issue for me. Also, the callback error happens if the device is too far. For me, RSSI values above -50 could connect reliably whereas RSSI values below that would consistently return the 133 error. Cheers!