dotnet-bluetooth-le/dotnet-bluetooth-le #264: iOS implementation of IAdapter does not invoke DeviceDisconnected on bluetooth turn off

d768, 2017-11-13:
iOS adapter does not invoke DeviceDisconnected event on bluetooth turn off.
#263 

## Steps to reproduce

1. Connect to BLE device

2. Turn bluetooth off

3. IAdapter.DeviceDisconnected is not invoked

## Expected behavior
iOS adapter should invoke DeviceDisconnected foreach Device on _deviceConnectionRegistry.

## Actual behavior
CBCentralState.UpdateState to CBCentralManagerState.PoweredOff is ignored and IAdapter.DeviceDisconnected is not invoked.

## Actual behavior
-

## Configuration

**Version of the Plugin:** e.g. 1.3.0

**Platform:** iOS 11.1

**Device:** e.g. iPhone 7

lamest, 2017-11-30:
same issue on iOS 11. looks like iOS 9 works fine.

xabre, 2017-12-10:
Merged the PR. Will add it to the next release

closed 2019-08-11 by xabre