dotnet-bluetooth-le/dotnet-bluetooth-le #266: Removed the delay(100) without any problems, resulted in speedup :)

Tijgerd, 2017-12-08:
I removed the delay from Characteristic.cs in line 127 and it fixed the delay between messages.
Or I fixed by editing something else.. can't remember..
But it sure didn't hurt the performance of my android application so that's a :+1: 

https://github.com/xabre/xamarin-bluetooth-le/blob/b30950113e40defcf1d612fdc972708be2137a16/Source/Plugin.BLE.Android/Characteristic.cs#L127

xabre, 2017-12-10:
- [ ]

Tijgerd, 2017-12-11:
I do need to note that I added a small delay after this function finished executing.
It seems the device needs to notice the notification request but I found that the delay after the execution of this function can be shorter (10ms)

xabre, 2019-08-11:
released with 2.1.0

closed 2019-08-11 by xabre