dotnet-bluetooth-le/dotnet-bluetooth-le #267: Incompatible using Code Sharing .NETStandard

JTrotta, 2017-12-09:
This plugin does not work with the new XPlatform Code sharing strategy template .NETStandard. 
The App can be compiled, but when this instruction is held, the App crashes:
`var _ble = CrossBluetoothLE.Current;`

Also in VS, got this warning:
[image]

# Steps to reproduce

1. Create a new blank application with the new template using code sharing template  .NETStandard

2. Add the plugin

3. Initialize Vanilla style

## Expected behavior
Working

## Actual behavior
Crash, calling `var _ble = CrossBluetoothLE.Current;`

### Crashlog

Unhandled Exception:
Java.Lang.ClassNotFoundException: <Timeout exceeded getting exception details> occurred

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
1.3.0
**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
Android 7.0
**Device:** e.g. HTC Sensation /i Phone 7 ...
Asus Zenphone 3 during debuging

jpflueger, 2017-12-09:
@JTrotta I'm able to use this library on both platforms without exception. Did you put this in your csproj file? 
```	
<PropertyGroup>
    <TargetFramework>netstandard2.0</TargetFramework>
    <AssertTargetFallback>$(AssertTargetFallback);portable45-net45+win8+wpa81</AssertTargetFallback>
</PropertyGroup>
```

JTrotta, 2017-12-10:
Hello. Yes I added it to the common project. But as soon as this instruction is called `CrossBluetoothLE.Current;` I got the java error "Java.Lang.ClassNotFoundException: occurred".
I'm using the very last version of VS17 15.5.1 and Android 4.8.0.753

dugasz1, 2017-12-15:
Same problem here with VS17 15.5.1 and Xamarin 4.8.0.753.

tribe84, 2017-12-28:
Same issue here.

biapar, 2018-01-30:
Solved?

xabre, 2018-06-15:
Starting with version 2.0.0 we target .net standard

closed 2018-06-15 by xabre