dotnet-bluetooth-le/dotnet-bluetooth-le #269: How to get the ScanResponse?

erdmenchen, 2017-12-13:
Maybe I am missing it in the documentation, but how can we get the scan response during 
`Adapter.StartScanningForDevicesAsync`?

In the device object that is reported in the `Adapter.DeviceAdvertised` Event  there does not seem to be any scan response related field...

xabre, 2017-12-16:
Check the 'AdvertisementRecords' proeprty of a device instance returned by the scan events.

From IDevice:
```
        /// <summary>
        /// All the advertisment records
        /// For example:
        /// - Advertised Service UUIDS
        /// - Manufacturer Specifc data
        /// - ...
        /// </summary>
        IList<AdvertisementRecord> AdvertisementRecords { get; }
```

erdmenchen, 2018-01-15:
So it looks like the `ScanResponse` field is automatically appended to the advertisement records by the currently used platform mechanics. Since the scan response carries its own data fields, they are found in the advertisement records each by its self, they are not bundled up as `ScanResponse` somehow.

I also misunderstanded/didn´t know that iOS / Android is already requesting the ScanResponse by its self and that it is appended to the advertisement records. I thought I have to somehow request it manually. 

Thanks!

closed 2018-01-15 by erdmenchen