dotnet-bluetooth-le/dotnet-bluetooth-le #273: Continuous connection question

shayo, 2018-01-09:
This is a question, not a real issue.
What are the rules regarding  ConnectToKnownDeviceAsync ?
When does it fail / time out?
How do I use the cancellation token to indicate to stop this operation?

I basically want to have a thread that continuously searches for a known device and tries to maintain connectivity with it. What's the best way to achieve this?
(i.e., try to connect, if fail after some predefined amount of time, try again).

lamest, 2018-02-02:
i do it in while(true) cicle, checking current connection state and connect if necessary

mullerch, 2018-02-14:
If your device is advertising when it accepts connection, I would check for the advertisement before starting the connection. This way you do not try if the device is not available.

lamest, 2018-02-15:
@mullerch Are you saying that ble device can be connected without advertising?

mullerch, 2018-02-15:
@lamest No I don't. Usually device do advertise when they are connectable and I think it's a good thing. I don't know the specification about that.

lamest, 2018-02-16:
ok, spec is that advertising is necessary to connection. So connection without discovery is same as with discovery but faster. Or this is how it appear to me

xabre, 2018-02-18:
Take a look at #145 maybe there's something that can help you there