dotnet-bluetooth-le/dotnet-bluetooth-le #274: Looking for maintainers

xabre, 2018-01-10:
If you are interested in being a maintainer let's start a conversation here :)

flyingxu, 2018-01-11:
Hi, I'd love to be one of them.

My day job is a Xamarin app using Bluetooth, I started using this package more than a year ago, call the APIs almost every day (but it's stable now so I don't debug into it often).
I tried porting the code to UWP (Some other's PR get merged but I got familiar with the code structures). I am planning to add MacOS support as well because it's needed by my day job.
I'm also a contributor of MvvmCross (accepted as a team member). This package introduced MvvmCross to me and then I spent more time there and use MvvmCross every day.

I'm not very active in the PRs here because I think it's quite stable now, at least for Android and iOS. 

This is a good package and it will help others if it gets maintained and adds more platform support.

xabre, 2018-01-15:
Hi @flyingxu, 
thank you for stepping up. We need some help with sorting through the open issues, maybe doing a few releases with the easy fixes. Great idea with the MacOS port, we have been wanting to add it ever since XF realease MacOS support which makes the sample app usable....but sadly not enough time. Doing the MacOS/UWP ports I think is a great way to get onboarded.

xabre, 2018-01-15:
I have just sent you an invite :)

flyingxu, 2018-01-15:
Thanks, I'll begin with the issues.

bed007, 2018-01-26:
Hi @xabre ,

I have been also using plugin-ble for quite a long time now and on several projects. I've also already contributed (once) to the project. I might have some time to contribute to the project. If you are interested in having extra hands, I would be happy to help!

jeevaeastpoint, 2018-02-07:
I would like to contribute as well

lucafabbri, 2018-02-09:
Glad to help!

bed007, 2018-02-23:
@xabre looks like there is few people that wants to step in this project. Would you consider adding more collaborators to the project ?

xabre, 2018-03-07:
@all Thanks for the support, from my point of view the more the merrier :)

xabre, 2018-03-07:
@bed007 I've just added you as a collaborator :) Glad to have you onboard

erdmenchen, 2018-03-21:
Since this project is not actively maintained, do you recommend any other xamarin forms ble plugin?

xabre, 2018-03-26:
@erdmenchen well this is not the right place for your question, if you would be interested in becoming a colaborator and 'actively' maintain the project then this is the right place for you :) and we welcome you with open arms, if not try stackoverflow.

pullelamurthy, 2018-05-13:
I can also help. I have been using this plugin for a long time and want to contribute to make this better. Please inform if there is anything I can do to help.

centrolutions, 2018-08-04:
How can I help get UWP support off the ground? Is there a starting point anywhere (i.e. the code that was tried and shelved)? I'd be happy to contribute to it since we have an existing project I'm interested in adding UWP support in.

flyingxu, 2018-08-06:
@centrolutions for UWP, you can see there is a branch of 'uwp_creators_update' at https://github.com/xabre/xamarin-bluetooth-le/branches

I tested and it work. The next step is to merge it into main branch

bugail, 2018-08-16:
@xabre I'd like to become a collaborator. Been using a plugin for awhile and have had great success with it. Been seeing a few issues with Android 8/9 and would love to help fix them :)

tfalasco, 2018-11-06:
Hello,
I would like to help maintain this project. A few years ago I wrote my own cross-platform BLE implementation for a Xamarin Forms app I wrote because I needed to support UWP, Android, and iOS. At the time this plugin didn't support UWP (plus I wanted to learn).
I like the work you're doing here and want to see it move forward. I think I can help out. Let me know if you're still looking for contributors. 
Thanks, 
Ted

xabre, 2018-11-16:
@bugail @tfalasco Nice to have you onboard.

JanNepras, 2019-02-08:
@xabre @flyingxu We are using your plugin on Android and iOS. We also need UWP and I would like to get it done. Could you add me as a contributor?

slimjim10, 2019-02-18:
I will be implementing Cancellation token for some calls which are currently not implemented, for example GetServiceAsync, GetCharacteristicAsync, ReadSync and WriteAsync.

I think this would be good to also merge into this project. How should I continue with this? Just do a pull request, or do you want to add me as a contributor?

jpark318, 2019-02-19:
@xabre I am currently using this wonderful package for my research team. Will be glad to help to maintain this plugin further, and I will also contribute to having ble part done for uwp as well! Could you add me as a contributor?

JanNepras, 2019-02-19:
@jpark318 I am going to work on the uwp probably next week. Could we work on this together?

jpark318, 2019-02-20:
> @jpark318 I am going to work on the uwp probably next week. Could we work on this together?

@JanNepras Yes, is there any documentation that would help me understand current progress on UWP? or should I just go through the code existing in uwp_creators_update branch??

MKuckert, 2019-03-12:
@xabre any chance to get on board also?

Thorinos, 2019-06-07:
I would like to be a collaborator too! I've send a pull request a few weeks ago and I would like to fix some issuse with the app demo. Let me know how I can help!