dotnet-bluetooth-le/dotnet-bluetooth-le #275: Question: how to cancel a connection request

shayo, 2018-01-15:
ConnectToKnownDeviceAsync(...)

Q: How can I cancel the connect request while it is still takes place?

erdmenchen, 2018-01-17:
Just use the `CancellationToken`, see https://github.com/xabre/xamarin-bluetooth-le/blob/18d04c5c5134b27e97dbdd19f1a4a94aeb122ed3/Source/Plugin.BLE.Abstractions/Contracts/IAdapter.cs#L114

shayo, 2018-01-17:
I thought CancellationToken is only to monitor the cancel operation. How do I initiate the cancel operation ?can you please post a small code snippet?
I didn’t find any Cancel method in the cancellationToken object.

erdmenchen, 2018-01-17:
You have to use a [CancellationTokenSource](https://msdn.microsoft.com/en-us/library/system.threading.cancellationtokensource).
There you have the `Token`-Property for the `ConnectToKnownDeviceAsync` call and also the [Cancel-Method](https://msdn.microsoft.com/en-us/library/dd321955) to indicate the cancellation.

e.g. (pseudo code):
```
var token = new CancellationTokenSource();
await adapter.ConnectToKnownDeviceAsync(new Guid(), token.Token);
token.Cancel();
```

shayo, 2018-01-17:
Gotcha. Thank you!

closed 2018-01-17 by shayo

erdmenchen, 2018-03-21:
I must confess, that I am experiencing issues with this approach on Android:

Triggering a connect and then cancelling it, does not stop the underlying connection process. 
Because later some time after cancellation via the cancellation token, suddenly the connection is established...