dotnet-bluetooth-le/dotnet-bluetooth-le #276: GattCallback error: 133 when want to connect device

kuropone, 2018-01-18:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Scan new device
ObservableCollection<IDevice> deviceList = new ObservableCollection<IDevice>();
            deviceList.Clear();

            //We have to test if the device is scanning 
            if (!ble.Adapter.IsScanning)
            {
                await adapter.StartScanningForDevicesAsync();

            }

            adapter.DeviceDiscovered += (s, a) =>
            {
                deviceList.Add(a.Device);
            };

2. Connect to selected device
Device.BeginInvokeOnMainThread(new Action(async () =>
            {
                try
                {
                    await CrossBluetoothLE.Current.Adapter.ConnectToDeviceAsync(device);
                }
                catch (DeviceConnectionException ex)
                {
                    await DisplayAlert("Notice", "BLE connect DCE ex msg: " + ex.Message, "OK");
                    await DisplayAlert("Name", "e.Device.Name = " + device.Name, "OK");
                    //send error message to screen if callback error 133 = "GattCallback error: 133"
                    //SendConnError(e.Device.Name);
                }
                catch (Exception ex)
                {
                    await DisplayAlert("Notice", "BLE connect basic ex msg: " + ex.Message, "OK");
                }
            }));

## Expected behavior
Tell us what should happen
Connected the device succesfully

## Actual behavior
Tell us what happens instead
GattCallback error: 133

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... Android 5.1.1 & Marshmallow

**Device:** e.g. HTC Sensation /i Phone 7 ... Xiaomi Redmi 3 & LG G3

lamest, 2018-02-02:
stop scaning before connect. and do this also on ui thread, android bluetooth is not threadsafe

bed007, 2018-02-06:
Yes, you must lock each writeasync/readasync operations so that only one execute at a time in Android, otherwise you get either Gatt 133 error or something like "WriteAsyncNative returned FALSE"

bed007, 2018-02-06:
I am using SemaphoreSlim.WaitAsync to perform locking

amentma, 2018-03-06:
can you provide sample code using SemaphoreSlim.WaitAsync locking? 
I'm also facing Gatt 133 when trying to connect to OBD II token with BLE. I have three different tokens, all of them are able to connect and communicate on iOS, but on Android I get 133 error when calling ConnectToDeviceAsync() on one specific device. Really need to support this peace, please help.

mjspk, 2019-01-01:
same problem here even when I call stopscanning first before call connect i still get this error GattCallBack error 133

epicstar, 2019-01-29:
I had this problem for a couple days. I fixed it with the following with my Pixel 2 XL:
```csharp
var connectParams = new ConnectParameters(true, true); // I need both autoconnect and ble transport set to true
await adapter.StopScanningForDevicesAsync();
await adapter.ConnectToDeviceAsync(device); // this call may take 10-20+ seconds... but only on Android
// do something else
```

xabre, 2019-08-11:
A serial queue would solve this issue for android (iOS does this implicitly already)

lamest, 2019-08-19:
> 
> 
> A serial queue / semaphore would solve this issue for android (iOS does this implicitly already)

It's not obvious, error prone, not crossplatform.

xabre, 2019-08-19:
@lamest That is a very good point. I'm ready to review your PR just can't seem to find it anywhere

stackunderflows, 2019-09-17:
I'm also getting this error. I'm performing a scan and populating a ListView as devices are found. The user is allowed to select a device before the scanning is complete, which initiates connecting to it. When this happens, I cancel the CancellationToken, call Adapter.StopScanningForDevicesAsync, and then attempt to connect to the device. This is when I get the 133 error. How can semaphores help this situation? Even if I implement my own queue, how can I tell that the Gatt is ready to accept the next command?

stackunderflows, 2019-09-17:
FYI - the sample app has this problem too.

xabre, 2019-09-17:
Release **2.2.0-pre2** offers a experimental implementation of a main thread serial queue (enabled for android only). 

All BLE android ble operations are covered except scanning as that can run for longer amounts of time and would block the queue.

Would appreciate some feedback ;)

unruledboy, 2020-02-02:
@xabre  searched the whole codebase and found the `BleCommandQueue` class. We currently have a problem with WriteAsync then by using ReadCharacterstic.ValueUpdated to get the result, but data comes sporadically, as in we send 10 commands, only getting 2-3 updates.

I am not sure by using BleCommandQueue will help this. And can we make better combination of the Write/Update with this helper class?

Bhauraj001, 2020-02-03:
@xabre  - It's not working properly in 2.2.0-pre2 version also, I downloaded and checked the issue. The issue is still exists..

AndrejBykadorov, 2020-04-20:
@xabre Good evening, is there some progress or workaround in this issue, please?

Storm-Develop, 2020-06-15:
Ok, so semaphores don't work on the main thread. Due to release before the value sent.
So here's my approach. It did work for me. 
### Serial queue.
```c#

private void WriteValue(string value)
        {

            try
            {
                if (_bluetoothSettingsViewModel.DeviceX1384.IsConnected && CharacteristicWriteSelected != null)
                {
                    var data = _asciiEncoder.GetBytes(value);
                    Device.BeginInvokeOnMainThread(async () =>
                    {
                        await writeQueue.QueueTask(() =>
                        {
                            CharacteristicWriteSelected.WriteAsync(data);
                        });
                    });

                }
                else
                {
                    _userDialogs.ErrorToast("Error", $"{_bluetoothSettingsViewModel.DeviceX1384.Name} is disconnected. Please reconnect", TimeSpan.FromSeconds(5));
                }

            }

            catch (Exception ex)
            {
                _userDialogs.ErrorToast("Error", "An error occured while sending data", TimeSpan.FromSeconds(5));

            }

        }

    public class BackgroundQueue
    {
        private Task previousTask = Task.FromResult(true);
        private object key = new object();

        public Task QueueTask(Action action)
        {
            lock (key)
            {
                previousTask = previousTask.ContinueWith(
                  t => action(),
                  CancellationToken.None,
                  TaskContinuationOptions.None,
                  TaskScheduler.Default);
                return previousTask;
            }
        }

        public Task<T> QueueTask<T>(Func<T> work)
        {
            lock (key)
            {
                var task = previousTask.ContinueWith(
                  t => work(),
                  CancellationToken.None,
                  TaskContinuationOptions.None,
                  TaskScheduler.Default);
                previousTask = task;
                return task;
            }
        }
    }
```
Reference: https://trenki2.github.io/blog/2019/01/02/background-queue-in-csharp/