dotnet-bluetooth-le/dotnet-bluetooth-le #278: StartScanningForDevicesAsync throw Exception

KhangYu, 2018-01-20:
Hi, guys,
I got an exception when try to scan BLE devices. The message showed "Field overflow on store 'InterpretedSystem.Runtime.ComplierServices.lAsyncStateMachineVT.<>U__1".
Does anyone have this issue before? Thanks :)

`IAdapter adapter = CrossBluetoothLE.Current.Adapter;
adapter.DeviceDiscovered += (s, a) =>  { };
adapter.StartScanningForDevicesAsync();`

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:** iOS 10.1

**Device:** iPhone 7+

lamest, 2018-01-22:
could it be recursion in your code that call async methods repeatedly?

KhangYu, 2018-01-23:
Actually i run my sample app on xamarin live player, could it be the cause of this issue?

xabre, 2019-08-05:
Try the sample app :)

closed 2019-08-05 by xabre