dotnet-bluetooth-le/dotnet-bluetooth-le #280: DeviceDisconnected is not called on clean peripheral disconnection (Gatt status code 19)

JeanBedard, 2018-01-26:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Establish a connection to a BLE peripheral device

2. Have the peripheral send a connection close event (Gatt Status 19 on Android)

3. The Adapter.DeviceConnectionLost event is raised

## Expected behavior
Since this is a clean disconnect (from the peripheral device), the GattCallback should rather raise the DeviceDisconnected event instead of the DeviceConnectionLost

## Actual behavior
The DeviceConnectionLost is raised. Hence, there is no way for the client to discriminate the reason of the connection lost, whether it is a signal lost because of a range/battery issue or whether it is a clean disconnect from the peripheral device.

### Crashlog

No exception thrown

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:** Android (all API level)

**Device:** Not relevant

bed007, 2018-01-26:
I am preparing a PR to resolve the issue.

flyingxu, 2018-01-27:
Is this android only? (I see the PR to fix it is android only)

Curiously to ask, how do you do the following? 

> Have the peripheral send a connection close event

I'd like to verify but I'd like to see how you reproduce the issue.

bed007, 2018-01-29:
Hi @flyingxu , I think this is only an Android but could be also tested on iOS. I will test it this week.

I've ask our embedded developer to send me the TI chip command to disconnect the BLE (on the peripheral side). I am pretty sure you can use tools like Nordic Semiconductor nRF tools to reproduce this also.

flyingxu, 2018-01-29:
Nice to know that, actually I have one Nordic Bluetooth USB toolkit.

bed007, 2018-01-29:
Nice! Btw, since you are there, I am trying to build the project using the build.bat file... I get compilation errors but when building from VS 2017, it builds & run fine. Any clue ?

flyingxu, 2018-01-29:
No idea about that, please share your result if you figure it out.

(I will delete your comment and mine some time later to keep the issue clean)

bed007, 2018-01-30:
@flyingxu : I confirm you that the issue also exists in iOS. I'll update the pull request accordingly once I have a fix.

Thanks
Jean

xabre, 2019-08-04:
Fixed by #281

closed 2019-08-04 by xabre