dotnet-bluetooth-le/dotnet-bluetooth-le #284: Crash when scanning for devices.

Etzix, 2018-02-06:
## Steps to reproduce

1. Scan for new devices

2. 50% of the time it will get an error when "DeviceDiscorvered" tries to add the device to a list.

## Expected behavior
DeviceDiscovered should add the device to a list.

## Actual behavior
It crashes with an exception.

### Crashlog

`02-06 13:36:47.655 I/MonoDroid(18466): UNHANDLED EXCEPTION:
02-06 13:36:47.705 I/MonoDroid(18466): Java.Lang.NullPointerException: println needs a message
02-06 13:36:47.705 I/MonoDroid(18466):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <657aa8fea4454dc898a9e5f379c58734>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at Java.Interop.JniEnvironment+StaticMethods.CallStaticIntMethod (Java.Interop.JniObjectReference type, Java.Interop.JniMethodInfo method, Java.Interop.JniArgumentValue* args) [0x00069] in <e736913786c2475188869561ae512b72>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at Java.Interop.JniPeerMembers+JniStaticMethods.InvokeInt32Method (System.String encodedMember, Java.Interop.JniArgumentValue* parameters) [0x00018] in <e736913786c2475188869561ae512b72>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at Android.Util.Log.WriteLine (Android.Util.LogPriority priority, System.String tag, System.String msg) [0x00059] in <8211f1f12f9e45838d631abf63e51c79>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at Enigma.Droid.BluetoothDeviceManager+<Scan>d__7.MoveNext () [0x00166] in C:\GIT\Passiq_App\Enigma\Enigma\Enigma.Android\BluetoothDeviceManager.cs:67 
02-06 13:36:47.705 I/MonoDroid(18466): --- End of stack trace from previous location where exception was thrown ---
02-06 13:36:47.705 I/MonoDroid(18466):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <657aa8fea4454dc898a9e5f379c58734>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <657aa8fea4454dc898a9e5f379c58734>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <657aa8fea4454dc898a9e5f379c58734>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <657aa8fea4454dc898a9e5f379c58734>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <657aa8fea4454dc898a9e5f379c58734>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at Enigma.Views.ConnectPage+<ListView_OnRefreshing>d__17.MoveNext () [0x0002c] in C:\GIT\Passiq_App\Enigma\Enigma\Enigma\Views\ConnectPage.xaml.cs:104 
02-06 13:36:47.705 I/MonoDroid(18466): --- End of stack trace from previous location where exception was thrown ---
02-06 13:36:47.705 I/MonoDroid(18466):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <657aa8fea4454dc898a9e5f379c58734>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) [0x00000] in <657aa8fea4454dc898a9e5f379c58734>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at Android.App.SyncContext+<>c__DisplayClass2_0.<Post>b__0 () [0x00000] in <8211f1f12f9e45838d631abf63e51c79>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at Java.Lang.Thread+RunnableImplementor.Run () [0x00008] in <8211f1f12f9e45838d631abf63e51c79>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00008] in <8211f1f12f9e45838d631abf63e51c79>:0 
02-06 13:36:47.705 I/MonoDroid(18466):   at (wrapper dynamic-method) System.Object:22a0d948-de84-438e-ae2d-e0fbe968fddb (intptr,intptr)
02-06 13:36:47.705 I/MonoDroid(18466):   --- End of managed Java.Lang.NullPointerException stack trace ---
02-06 13:36:47.705 I/MonoDroid(18466): java.lang.NullPointerException: println needs a message
02-06 13:36:47.705 I/MonoDroid(18466): 	at android.util.Log.println_native(Native Method)
02-06 13:36:47.705 I/MonoDroid(18466): 	at android.util.Log.println(Log.java:429)
02-06 13:36:47.705 I/MonoDroid(18466): 	at mono.java.lang.RunnableImplementor.n_run(Native Method)
02-06 13:36:47.705 I/MonoDroid(18466): 	at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:30)
02-06 13:36:47.705 I/MonoDroid(18466): 	at android.os.Handler.handleCallback(Handler.java:733)
02-06 13:36:47.705 I/MonoDroid(18466): 	at android.os.Handler.dispatchMessage(Handler.java:95)
02-06 13:36:47.705 I/MonoDroid(18466): 	at android.os.Looper.loop(Looper.java:157)
02-06 13:36:47.705 I/MonoDroid(18466): 	at android.app.ActivityThread.main(ActivityThread.java:5293)
02-06 13:36:47.705 I/MonoDroid(18466): 	at java.lang.reflect.Method.invokeNative(Native Method)
02-06 13:36:47.705 I/MonoDroid(18466): 	at java.lang.reflect.Method.invoke(Method.java:515)
02-06 13:36:47.705 I/MonoDroid(18466): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1265)
02-06 13:36:47.705 I/MonoDroid(18466): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1081)
02-06 13:36:47.705 I/MonoDroid(18466): 	at dalvik.system.NativeStart.main(Native Method)
02-06 13:36:47.715 D/Mono    (18466): DllImport searching in: '__Internal' ('(null)').
02-06 13:36:47.715 D/Mono    (18466): Searching for 'java_interop_jnienv_throw'.
02-06 13:36:47.715 D/Mono    (18466): Probing 'java_interop_jnienv_throw'.
An unhandled exception occured.

02-06 13:36:47.715 D/Mono    (18466): Found as 'java_interop_jnienv_throw'.
`

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:**  Android 4.4 / Android 7.0

**Device:** Samsung Glaxy Note 3, HTC(unknown)

bed007, 2018-02-06:
Can you post a sample of your code with little context around it ? I never had this exception before.

Etzix, 2018-02-07:
This was fixed by forcing it to scan on the mainthread instead. I believe the reason its throwing that error might be because its missing an error message for that in the code somewhere.

closed 2018-02-07 by Etzix