dotnet-bluetooth-le/dotnet-bluetooth-le #285: .NETStandard v2.0 support

dinodzakmic, 2018-02-23:
By creating new Xamarin.Forms project based on .NETStandard 2.0, Plugin.BLE packages are restored using .NETFramework v4.6.1. 

Rebuilding the solution inside VisualStudio 2017 is successfull, but if I try to set up Jenkins or VS Mobile center for CI/CD/CT, then rebuild fails because of this "issue".
Inside Jenkins I am trying to build using MSBUILD.

Is there any workaround or do you soon plan to add .NETStandard 2.0 support?

biapar, 2018-02-28:
Do it work with .NETStandard 2.0?

dinodzakmic, 2018-03-06:
@biapar  It's working when I use my machine for build and deployment, but on my Jenkins server it does not work.

biapar, 2018-03-06:
Which is the min .NET version supported?

smstuebe, 2018-04-04:
its using PCL profile 259 which is netstandard1.0. netstandard1.0 is a subset of netstandard2.0. So it will work.

flyingxu, 2018-06-23:
Plugin.BLE (2.0.0-pre1) is .NETStandard 1.0
```
NETStandard.Library (>= 1.6.1) 
```

closed 2018-06-23 by flyingxu