dotnet-bluetooth-le/dotnet-bluetooth-le #287: Question. How to use example app ?

Anandks1993, 2018-03-22:
I am newbie to Xamarin and I want to try the example app provided by this plugin. But I don't have an idea how to implement the example. Can someone pls explain how to try out the example app. Thanks is advance 🤝

flyingxu, 2018-04-03:
The best first step to try this is to build the sample from source. You can see there is 1 sample for andoid and 1 for iOS in https://github.com/xabre/xamarin-bluetooth-le/tree/master/Source/BLE.Client .

 	BLE.Client.Droid 	
	BLE.Client.iOS

closed 2018-04-03 by flyingxu

ikebarrett, 2018-06-17:
I downloaded and tried the sampl, set configuration to deploy droid.  I get a system.nullreference set.  There is no indication as to why.  Any thoughts on what I am doing wrong?

hatefk, 2021-07-16:
@flyingxu my understanding is that we have to write two sets of code to get BLE going for both Android and iOS. Am I missing some point or do I need two separate codes for cross platform apps?