dotnet-bluetooth-le/dotnet-bluetooth-le #296: [iOS] Cannot unsubscribe from ValueUpdated event

Yelnik, 2018-04-20:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

This happens even after getting the characteristics again

## Steps to reproduce

1. Subscribe to `ValueUpdated` event

2. Unsubscribe from event, and re-subscribe

3.

## Expected behavior
I expect the event to fire once

## Actual behavior
The event fires twice, on the same thread

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:** e.g. iOS 10.3 and iOS 11.3

**Device:** e.g. iPhone 5C, iPhone 7

coltonf93, 2018-11-08:
I'm also experiencing this issue on Android (5.0, 6.0, 7.0, 8.0, 8.1, 9.0)
Has anyone found a workaround for this yet?

coltonf93, 2018-11-11:
The problem I was running into was that after disconnecting and reconnecting the same device a new IDevice instance was created, meaning new service and characteristic instances were created as well. By caching the last instance of the characteristic I was able to unsubscribe successfully:

`private ICharacteristic _lastSubscribedCharacteristic;

private async Task<bool> SubscribeDeviceTransferServices()
{
        var someTransferService = await this.DeviceWrapper.NativeDevice.GetServiceAsync(SOME_SERVICE_UUID);
        if (someTransferService != null)
        {
            var someTransferCharacteristic = await someTransferService.GetCharacteristicAsync(SOME_CHARACTERISTIC_UUID);
            if (someTransferCharacteristic != null)
            {
                if (_lastSubscribedCharacteristic != null)
                {
                    //await _lastSubscribedCharacteristic.StopUpdatesAsync(); This causes issues
                    _lastSubscribedCharacteristic.ValueUpdated -= PacketReceivedHandler;
                }
                await someTransferCharacteristic.StartUpdatesAsync();
                someTransferCharacteristic.ValueUpdated += PacketReceivedHandler;
                _lastSubscribedCharacteristic = someTransferCharacteristic;
                return true;
            }
        }
    return false;
}`

A better approach may be to unsubscribe when the device disconnects rather than when the next device is attempting to connect. The above was just a proof of concept.