dotnet-bluetooth-le/dotnet-bluetooth-le #299: ReadAsync appears to ignore cancellation token

bwenninger, 2018-04-24:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Our application is reading data from another device and is doing so through calls to ReadAsync().  When the connection gets lost (we are notified via CrossBluetoothLE.Current.Adapter.DeviceConnectionLost we are calling StopUpdatesAsync on the read characteristic.  This didn't appear to be actually stopping the read so we switched to passing a cancellation token to ReadAsync and then calling cancel on that.  However this cancellation appears to be ignored.  In looking a bit at the source code it looks like the cancellation token is never passed on to the native read.

2.

3.

## Expected behavior
Tell us what should happen
I was expecting to be able to stop/cancel a read.
## Actual behavior
Tell us what happens instead
The read doesn't actually stop/cancel.
### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
1.0.3
**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
iOS 10.3.3
**Device:** e.g. HTC Sensation /i Phone 7 ...
iPad

spenceroo, 2018-05-11:
We are having the same issue trying to cancel a ReadAsync.  Anyone have ideas

Xiaoliangzhang, 2018-07-20:
We had the same problem for both ReadAsync and WriteAsync. Having looked into source code, it seems the cancellation tokens passed in are never used in either CharacteristicBase or platform specific implementation of Characteristic.

sorenhl, 2019-02-19:
> We had the same problem for both ReadAsync and WriteAsync. Having looked into source code, it seems the cancellation tokens passed in are never used in either CharacteristicBase or platform specific implementation of Characteristic.

Did you find a solution for this?

TwoTenPvP, 2020-12-28:
Still no solution for this?

I am actually in quite big need of this, might submit a PR unless you guys are working on it.