dotnet-bluetooth-le/dotnet-bluetooth-le #300: why does the device disconnect after 30 seconds?

urmatgit, 2018-04-28:
why does the device disconnect after 30 seconds? 
1) connect to the device 
2) connect to notify 
then about 30 seconds, the connection disconnect spontaneously 

## Expected behavior
spontaneously disconnects 

## Actual behavior
must operate by manually disconnect.

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:**  1.3.0

**Platform:** iOS 11.3

**Device:** iPhone 6s

alexcoupal, 2018-06-05:
@smsissuechecker  Any update ? I have the same issue.

pirhano, 2019-04-24:
Any updates on the issue ?

mijmijmijmij, 2020-10-26:
I am having the same issue, has anyone resolved this?

xabre, 2020-10-28:
Does this happen with the sample app as well?

mijmijmijmij, 2020-10-29:
Hi, I a determined the cause of the issue here for me, it was not related to the app at all but our BLE device. iOS has set connection parameters that we were not following, once updating our parameters, the issue was resolved.

https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

[image]

AmandaLou, 2021-03-24:
I was also experiencing this behaviour. I would connect to a characteristic and start receiving updates, but would be disconnected with no reason after around 30 seconds. I found that by pausing a bit between connecting to the characteristic and starting updates, my problem was solved. I'm not sure why that is, but this is what I'm using: 

`            ICharacteristic characteristic = await service.GetCharacteristicAsync(characteristicId);`
`           Thread.Sleep(500);`
`           characteristic.ValueUpdated += Characteristic_ValueUpdated;`
`           Thread.Sleep(1500);`
`            await characteristic.StartUpdatesAsync();`

marciosouzajunior, 2022-03-25:
I had the same issue and I found that this only happen when device is not bonded/paired with phone. I couldn't run the sample app (due to some errors), but I tested the device with "nRF Connect" app and confirmed the disconnection after 30 seconds. So it seems to be an iOS issue. Anyway, I found a workaround by reconnecting device every time it gets disconnected. I would appreciate if @xabre agree with this:

```
public class BLEReconnectTest {  

	private IBluetoothService bluetoothService;
	private ICharacteristic myCharacteristic;

	public BLEReconnectTest(IBluetoothService bluetoothService)
	{
		this.bluetoothService = bluetoothService;
		bluetoothService.DeviceDisconnected += DeviceDisconnected;
		bluetoothService.DeviceConnected += DeviceConnected;
	}

	private async void DeviceConnected(object sender, DeviceEventArgs e)
	{
		await StartUpdate();
	}

	private void DeviceDisconnected(object sender, DeviceEventArgs e)
	{
		StopUpdate();
		AttemptReconnect();
	}

	private async void AttemptReconnect()
	{
		try
		{
			await bluetoothService.ConnectToDevice(Profile.DispositivoId);
		}
		catch (Exception)
		{
			AttemptReconnect();
		}
	}

	public async Task StartUpdate()
	{
		myCharacteristic = await bluetoothService.GetHeartRateCharacteristic("myDeviceId");
		myCharacteristic.ValueUpdated -= ValueUpdated;
		myCharacteristic.ValueUpdated += ValueUpdated;
		await myCharacteristic.StartUpdatesAsync();
	}

	public void StopUpdate()
	{
		myCharacteristic.StopUpdatesAsync();
		myCharacteristic.ValueUpdated -= ValueUpdated;
	}

	private void ValueUpdated(object sender, CharacteristicUpdatedEventArgs args)
	{
		// Get value from args.Characteristic.Value[1];            
	}
}
```