dotnet-bluetooth-le/dotnet-bluetooth-le #306: i can find my device ,but i can't connect to the device when i use the method ConnectToDeviceAsync

kobin97, 2018-05-17:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.

2.

3.

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

kobin97, 2018-05-17:
public HeaderControlSample()
        {
            InitializeComponent();

var ble = CrossBluetoothLE.Current;
            var adapter = CrossBluetoothLE.Current.Adapter;

            adapter.DeviceConnected += Adapter_DeviceConnected;

            adapter.DeviceDiscovered += (s, a) =>
            {
                if (a.Device.Name != null)
                {
                    devicelist1.Add(a.Device);
                 
                }

            };

            await adapter.StartScanningForDevicesAsync();
            
            lv_deviceList.ItemsSource = devicelist1;

        }

 async public void OnSelection(object sender, SelectedItemChangedEventArgs e)
        {

            var DeviceSelect = e.SelectedItem as IDevice;
            var answer = await DisplayAlert("Device Connection", "Would you like to connect with this device:" + DeviceSelect.Name.ToString() + "?", "Yes", "No");

            if (answer)
            {
                try
                {
                    await CrossBluetoothLE.Current.Adapter.StopScanningForDevicesAsync();

                    await CrossBluetoothLE.Current.Adapter.ConnectToDeviceAsync(DeviceSelect);
                }
                catch (DeviceConnectionException ex)
                {
                    System.Diagnostics.Debug.WriteLine("EXCEPTION\nDevice : " + ex.ToString());
                }
            }

        }

kobin97, 2018-05-17:
when is use the method "ConnectToDeviceAsync",nothing happen,the device state is always "Disconnected"

jeevaeastpoint, 2018-05-22:
Have you checked whether "DeviceConnected" event is fired on connecting to the device?

xabre, 2019-08-05:
Not reproducible with sample app. Not enough information. Reopen issue if still a problem and add details.

closed 2019-08-05 by xabre