dotnet-bluetooth-le/dotnet-bluetooth-le #309: How to handle GATT write errors more efficiently

Rousseau38, 2018-06-01:
Hello, 

How do you guys handle, resolve Gatt writeasync errors with this package? I read earlier posts and followed android guidelines (don't scan while connected, write on the main thread), but I still receive the errors. 

Here is how I writeasync: 

    await OnMainThreadAsync(async () =>
                                    {
                                        await device.characteristic.WriteAsync(array);                                                      
                                    });

Here is OnMainThreadAsync: 

        public static async Task OnMainThreadAsync(Func<Task> action)
        {
            var tcs = new TaskCompletionSource<object>();
            Xamarin.Forms.Device.BeginInvokeOnMainThread(async () => {
                try
                {
                    await action();
                    tcs.SetResult(new { });
                }
                catch (Exception ex)
                {
                    tcs.SetException(ex);
                }
            });
            await tcs.Task;
        }

Thanks

xabre, 2019-08-05:
Also make sure to queue your write requests especially on Android. Several concurrent write requests will lead to GATT 133 errors.

closed 2019-08-05 by xabre

Umer-Mahmood, 2021-04-22:
> Also make sure to queue your write requests especially on Android. Several concurrent write requests will lead to GATT 133 errors.

How to queue?