dotnet-bluetooth-le/dotnet-bluetooth-le #310: Bluetooth device is not getting disconnected

parithi4u, 2018-06-04:
I am developing Xamarin android native application where to connect the Bluetooth devices  using Plugin.BLE 1.3.0 version.  Actually our application should have the following features that connect  multiple Bluetooth devices one by one and disconnect those devices one by one.

The beauty is, we are able  to connect the device one by one  as expected,but we are getting issues while disconnecting the device.   Assume that I have 5 Bluetooth devices where I have tried to connect  one by one then disconnecting the device one by one.  
Most of times it was successful , but few times the Bluetooth device is not getting disconnected.

Could you please suggest that how to handle the disconnect device ?

To disconnect the event , I am calling the below functions:
`await CrossBluetoothLE.Current.Adapter.DisconnectDeviceAsync(device);`

## Configuration

**Version of the Plugin:** Plugin.BLE
 version="1.3.0" targetFramework="portable46-net451+win81"

**Platform:** : Android API 21

**Device:**  Samsung Tab S3 , Google Pixel XL, Samsung Galaxy J7 Prime (Android 7.0)

asierdanjou, 2018-12-07:
Same Issue.
Galaxy S4 with android 5.0

Steps:
1- device = await adapter.ConnectToKnownDeviceAsync(Guid.Parse("00000000-0000-0000-0000-" + dispo.Direccion));
2-                    
try { await adapter.DisconnectDeviceAsync(device); }  catch (DeviceConnectionException ex)                    {                        await DisplayAlert("Error", ex.Message.ToString(), "Accept");                    }
 
Expected behavior:
My Silicon Labs ble module gets disconnected.

Actual behavior: 
"Try disconnect" occurs successful but peripheral ble module is still connected. I think Limited state.

This issue occurs sometimes.

nicktiberi, 2019-09-10:
We have encountered this in our app (on Android only), and I believe it was because we were accidentally trying to connect to the same device multiple times. After adding logic into our app to detect whether a device was already connected, this problem seems to have gone away. Fingers crossed!

```
await adapter.ConnectDeviceAsync(device); //connect to device first time
//do some stuff
await adapter.ConnectDeviceAsync(device); //connect to device again
//do some more stuff
await adapter.DisconnectDeviceAsync(device); //disconnect from device
//at this point device is stuck in limited state