dotnet-bluetooth-le/dotnet-bluetooth-le #321: BLE does not scan

bleari, 2018-08-07:
## Steps to reproduce

1. Plugin has been added as suggested.

2. Visual studio and Xamarin.Forms are updated to the latest version.

3. Code attached below
```
using System;

using Android.App;
using Android.Content.PM;
using Android.Runtime;
using Android.Views;
using Android.Widget;
using Android.OS;
using Plugin.BLE;
using Plugin.BLE.Abstractions.Exceptions;
using System.Collections.Generic;
using Android.Bluetooth;
using Whale_Watch.Droid;

namespace Whale_Watch.Droid
{
    [Activity(Label = "BLE_Test",  Theme = "@style/MainTheme", MainLauncher = true, ConfigurationChanges = ConfigChanges.ScreenSize | ConfigChanges.Orientation)]
    public class MainActivity : global::Xamarin.Forms.Platform.Android.FormsAppCompatActivity
    {
        protected async override void OnCreate(Bundle bundle)
        {
            TabLayoutResource = Resource.Layout.Tabbar;
            ToolbarResource = Resource.Layout.Toolbar;

            base.OnCreate(bundle);

            global::Xamarin.Forms.Forms.Init(this, bundle);
            LoadApplication(new App());
            await scanAsync();
        }
        public async System.Threading.Tasks.Task scanAsync()
        {          
            List<Plugin.BLE.Abstractions.Contracts.IDevice> deviceList = new List<Plugin.BLE.Abstractions.Contracts.IDevice>();
            var ble = CrossBluetoothLE.Current;
            var adapter = CrossBluetoothLE.Current.Adapter;
            var state = ble.State;           
            adapter.ScanTimeout = 200000;
            adapter.DeviceDiscovered += (s, a) => deviceList.Add(a.Device);
            await adapter.StartScanningForDevicesAsync();            
        }      
    }
}
```
## Expected behavior
It should scan for all nearby Bluetooth le devices.

## Actual behavior
It will run the code and no errors are formed, however even though there are ble devices visible it does not return any or add them to the list. I am unsure if my code is at fault.

### Crashlog

Does not crash or cause an exception

## Configuration

**Version of the Plugin:** 1.3.0

**Platform:** Compiled using Android 8.1

**Device:** Moto E4 Plus, but will also occur on all other devices tested.

flyingxu, 2018-08-13:
Could you test with the test client (Source\Ble.Client) ?

samosim, 2018-08-14:
Did you add permissions?

martinik91, 2018-10-02:
Hey, I just upgraded the target framework for one of our project to 8.1 and started experiencing the exact same issue. Downgrading to 8.0 seems to fix the problem so we'll just keep it like that for now.
I'm also on version 1.3.0.

martinik91, 2018-10-03:
So, I did some more testing today, and I can't get the Source\BLE.Client to discover my bluetooth devices at all. Tried 7.1, 8.0 and 8.1.

For our own project, everything seems to be working fine using SDK 8.0. I tried both 8.1 and 9.0 without any luck. No errors as in the original issue, but no BLE devices are discovered. When running in 8.0 I discover the device I'm interested in immediately in addition to a bunch of other devices in the building. Could there be some permission changes or similar for 8.1 and 9.0?

obtpaul, 2018-10-04:
I've been chasing this same problem for days! The same app BLE.Client runs correctly. It is difficult to follow, given it uses the raw code and not the plugin. I've also used the sample code from https://github.com/didourebai/BLEPluginDemo that uses this same plugin with the same problem. No compile or runtime errors, just never discovers a device. All updates applied. All permissions applied. I've tried the rollback of Target SDK, but that didn't help.

martinik91, 2018-10-16:
@StriBog45 Did the answer here work for you?

I tried to implement permission checking for Android, but ContextCompat.CheckSelfPermission reports that all my permissions are already granted. Still no results from scanning..

martinik91, 2018-10-16:
I've done even more digging, and found some really strange results.
TL;DR workaround for now: Remove the android:targetSdkVersion from the uses-sdk tag in your AndroidManifest.xml and everything works. Full tag from our project:
`<uses-sdk android:minSdkVersion="15" />`

Long story:
Our manifest file didnt have any android:targetSdkVersion property in the uses-sdk tag, and everything was working fine (still is if I don't add it)

Whenever I change the Target Framework in the Android Project properties window (under application), the uses-sdk tag is automatically updated to include the android:targetSdkVersion property. For some reason this causes scanning to stop working like in the original issue.

After inspecting the code, I suspected the plugin was using the old scan implementation for Adapter.cs in the Android project from line 66:

```
if (Build.VERSION.SdkInt < BuildVersionCodes.Lollipop)
{
    StartScanningOld(serviceUuids);
}
else
{
    StartScanningNew(serviceUuids);
}
```

.. But no dice, it runs the "StartScanningNew" function in all my test cases, and it only works if the android:targetSdkVersion is removed from the manifest. I've been stepping through the code multiple times, and it all runs in the exact same way, except the ScanCallback implementation is never called when targetSdkVersion is present.

I don't really know enough about sdk versions etc. to test this further right now, but I'll read some more into it tomorrow.

It seems to me that the app is built with the sdk set in Target Framework, since Build.VERSION.SdkInt reports P (28) for me now. But during startup I get a Warning stating that the app was built for an older version of the OS. I think this is something new introduced in P(?).

martinik91, 2018-10-16:
Okay, so this is pretty frustrating. Apparently you have to enable Location services for this to work on newer OS versions. Turn on location and everythign works fine no matter the version or manifest!

Found this out from a stackoverflow article, but haven't seen ANYTHING about it in the android or Xamarin docs... Hope this solves some more people's issues as well! Maybe a note about this should be added to the readme here as well?

obtpaul, 2018-10-16:
I fixed my scan and connect issues by adding the following code to Android MainActivity.cs.
Add this line in the OnCreate method:

` Plugin.CurrentActivity.CrossCurrentActivity.Current.Init(this, savedInstanceState);`

And then add this OnRequestPermissionsResult override:
`public override void OnRequestPermissionsResult(int requestCode, string[] permissions, [GeneratedEnum] Android.Content.PM.Permission[] grantResults)
        {
            PermissionsImplementation.Current.OnRequestPermissionsResult(requestCode, permissions, grantResults);            
        }`

I followed everything else I could find, but this fixed it.

muhammetsahin, 2019-01-22:
I have same problem. I tried this (https://julianocustodio.com/2017/12/07/bluetooth/) demo app but not working.

lucadello91, 2019-06-20:
Has anyone solved this problem? Adding and requesting permissions does not seem to work.
The @ghost solution works, but it prevents me from loading the apk on the Play Store.
How can I do?

xabre, 2019-08-04:
Please check the documentation and sample project.

closed 2019-08-04 by xabre

janseris, 2023-03-14:
> Okay, so this is pretty frustrating. Apparently you have to enable Location services for this to work on newer OS versions. Turn on location and everythign works fine no matter the version or [...]

This still applies. Very frustrating. 
Android 12